ros-planning/navigation #617: CMake Error

dahlan77, 2017-08-22:
When i cmake navigation with ros indigo 

I got some errors that i havent met before

how to repair this errors?

 Using these message generators: gencpp;genlisp;genpy
CMake Error at /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:108 (message):
  Could NOT find Bullet (missing: BULLET_DYNAMICS_LIBRARY
  BULLET_COLLISION_LIBRARY BULLET_MATH_LIBRARY BULLET_SOFTBODY_LIBRARY
  BULLET_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/FindPackageHandleStandardArgs.cmake:315 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-2.8/Modules/FindBullet.cmake:76 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  navigation/map_server/CMakeLists.txt:11 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/dahlan/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dahlan/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1

best regard
Dahlan
Bandung Institute Technology

DLu, 2017-08-28:
It looks like you don't have the proper dependencies for the navigation stack. Use `rosdep` to install them. http://wiki.ros.org/ROS/Tutorials/rosdep

closed 2017-08-28 by DLu

DLu, 2018-05-22:
Have you run `rosdep install navigation`?

pouyx, 2018-05-22:
you can try this Link sudo apt-get install libbullet-dev

zakiryne, 2019-02-08:
did you solve this issue? if yes how, I have the same problem.

imanpakii, 2019-03-26:
@zakiryne 
Hello,
I have ROS indigo,
I solved it with these steps: 
1- I install the indigo-devel branch instead of master. It seems the master for different ROS distro:
`git clone -b indigo-devel https://github.com/ros-planning/navigation.git`
and then I got this error:
`-- checking for module 'orocos-bfl' -- package 'orocos-bfl' not found CMake Error at /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:283 (message): A required package was not found Call Stack (most recent call first): /usr/share/cmake-2.8/Modules/FindPkgConfig.cmake:337 (_pkg_check_modules_internal) navigation/robot_pose_ekf/CMakeLists.txt:6 (pkg_check_modules)`
then I install binary of navigation and it's fixed the problem as it said [https://answers.ros.org/question/201630/orocos-bfl-not-found-while-installing-navigation-stack-ros-indigo-ubuntu-1404/](url)
2- `sudo apt-get install ros-indigo-navigation`

yunjc, 2020-06-06:
> you can try this Link sudo apt-get install libbullet-dev

Solved with this. Thank you!