ros-planning/navigation #618: move_base_msgs error

dahlan77, 2017-08-22:
i installed with ros indigo

and i compiled it and meet some errors

anyone can suggest to me to fix this error

 +++ processing catkin package: 'simple_navigation_goals'
-- ==> add_subdirectory(simple_navigation_goals)
-- Using these message generators: gencpp;genlisp;genpy
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  simple_navigation_goals/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'move_base_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "move_base_msgs"
  with any of the following names:

    move_base_msgsConfig.cmake
    move_base_msgs-config.cmake

  Add the installation prefix of "move_base_msgs" to CMAKE_PREFIX_PATH or set
  "move_base_msgs_DIR" to a directory containing one of the above files.  If
  "move_base_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  simple_navigation_goals/CMakeLists.txt:7 (find_package)

DLu, 2017-08-28:
Please use answers.ros.org for this type of inquiry, as the problem sounds like it originates in your `simple_navigation_goals` package and not move_base.

closed 2017-08-28 by DLu

Napoleon-BlownApart, 2020-02-15:
What happened to the `move_base_msgs` package?  It seems to be missing in the root list of sub-packages here.
I've cloned this `navigation` repo as part of my ROSPlan demo(Stage Exploration) build. I can install it via `apt` but prefer to build from source.
Where can I get current `move_base_msgs` source for `melodic`?

mikeferguson, 2020-02-15:
We've moved move_base_msgs into a separate repo (the navigation_msgs you mention above), so that people interacting with move_base don't need to build all of move_base locally. The jade-devel branch should work on melodic (we have not changed the message since the package split, which was done for Jade).

The wiki page you reference is up to date - in the "Package Summary" for the "Melodic" page, it tells you the repository listed above and jade-devel is the branch to use for source.