ros-planning/navigation #635: Support in map_server for Rather Large Maps

DLu, 2017-11-16:
As reported [here](https://groups.google.com/forum/#!topic/ros-sig-navigation/ekGOmMAMtjM)

> I need to localize a robot in a warehouse of 415x270 meters with a <3cm precision, which make a map of 20768x13612px at 0.020 m/px and a pgm file of 282 MB. (Or 1.7 MB in png).
I tried to load it to AMCL but map_server cannot even load the map on a PC with 16GB of ram because SDL yell "Out of memory". I manage to load it by cropping some part of the map, but that's not a real solution. I could also lower the resolution, but it would be a temporary fix until I need a bigger map.

mikeferguson, 2018-09-10:
I think this ticket was solely created from the discussion -- AFAIK, there is nobody actively working on a fix specifically for this. Separately, there is a ticket about changing the datatypes in AMCL -- https://github.com/ros-planning/navigation/issues/778 -- which also has an active PR -- https://github.com/ros-planning/navigation/pull/791 -- you might give that a try.

SteveMacenski, 2018-09-10:
For a given application, I know plenty of companies that tile their maps and load only sections at a time that are relevant in order to help deal with this. It's also a common strategy used in 3D dense and sparse reconstruction due to the amount of data stored.

eduidl, 2021-06-01:
This issue is due to restriction in SDL.

https://github.com/libsdl-org/SDL-1.2/blob/main/src/video/SDL_surface.c#L47-L52

```
	/* Make sure the size requested doesn't overflow our datatypes */
	/* Next time I write a library like SDL, I'll use int for size. :) */
	if ( width >= 16384 || height >= 65536 ) {
		SDL_SetError("Width or height is too large");
		return(NULL);
	}
```

closed 2025-06-04 by mikeferguson