ros-planning/navigation #638: Failed to create the global_planner/GlobalPlanner planner

ryanoldja, 2017-11-30:
on the TurtleBot platform, when launching the `move_base` node configured for the global_planner/GlobalPlanner, I receive the following error:

`Failed to create the global_planner/GlobalPlanner planner, are you sure it is properly registered and that the containing library is built? Exception: According to the loaded plugin descriptions the class global_planner/GlobalPlanner with base class type nav_core::BaseGlobalPlanner does not exist. Declared types are  navfn/NavfnROS`

I load `move_base_params.yaml` within `move_base` as follows:

` <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen" >`
`     <rosparam file="$(find robust_navigation)/param/move_base_params.yaml" command="load" />`

where the `base_global_planner` param is defined in `move_base_params.yaml` as follows:
`base_global_planner: "global_planner/GlobalPlanner"`

I've re-installed the nav stack to no avail.

DLu, 2017-11-30:
Please do not cross-post your questions. [answers.ros.org](https://answers.ros.org/question/276939/failed-to-create-the-global_plannerglobalplanner-planner/)

closed 2017-11-30 by DLu

markdole, 2018-11-19:
Can you please leave the right way to solve this problem? 
I met the same issue, and dont actually understand how to deal with the source problem.

usmanNoor5, 2025-01-25:
same issue and didn't get the solution till now