ros-planning/navigation #645: Local Costmap Does not Refresh when the Robot is Stationary.

sradmard, 2017-12-07:
Hi,
I am encountering the issue that my local cost map does not refresh when the robot is stationary. This would create significant problems when there are moving obstacles around the robot. You can observe the behaviour here at 0:22
[Local_costmap_issue](https://youtu.be/sHPK6NG8IzQ
)
Here is how I launch my move_base node in my robot launch file:
```
    <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
	<rosparam file="$(find turtlebot_vrep)/config/costmap_common_params.yaml" command="load" ns="global_costmap" />
	<rosparam file="$(find turtlebot_vrep)/config/costmap_common_params.yaml" command="load" ns="local_costmap" />
	<rosparam file="$(find turtlebot_vrep)/config/local_costmap_params.yaml" command="load" />
	<rosparam file="$(find turtlebot_vrep)/config/global_costmap_params.yaml" command="load" />
	<rosparam file="$(find turtlebot_vrep)/config/base_local_planner_params.yaml" command="load" />
	<rosparam file="$(find turtlebot_vrep)/config/move_base_params.yaml" command="load" />
	<remap from="cmd_vel" to="cmd_vel_mux/input/navi"/>
    </node>
```
Relevant data are getting published on /map topic at about 16 Hz, on /scan topic at 20 Hz, on /odom topic at 20 Hz, and on /tf topic at about 300 Hz. 
However, the "/move_base/local_costmap/costmap" topic barely gets published on, and does not publish anything when the robot is stationary. 

Also, here are my parameters:

> controller_frequency: 20.0
> TrajectoryPlannerROS:
>   max_vel_x: 10.45
>   min_vel_x: 0.1
>   max_vel_theta: 10.0
>   min_in_place_vel_theta: 0.4
>   acc_lim_theta: 3.2
>   acc_lim_x: 2.5
>   acc_lim_y: 2.5
>   holonomic_robot: true
> global_costmap:
>   global_frame: map
>   robot_base_frame: base_link
>   update_frequency: 1.0
>   static_map: true
>   publish_frequency: 0.5
>   transform_tolerance: 0.5
> local_costmap:
>   global_frame: map
>   robot_base_frame: base_link
>   update_frequency: 5.0
>   publish_frequency: 2.0
>   static_map: false
>   rolling_window: true
>   width: 3.0
>   height: 3.0
>   resolution: 0.05
> Move base node parameters.
> shutdown_costmaps: false
> controller_frequency: 5.0
> controller_patience: 3.0
> planner_frequency: 1.0
> planner_patience: 5.0
> oscillation_timeout: 10.0
> oscillation_distance: 0.2

I get the same behaviour on the actual turtlebot as well, and I would really appreciate any help,

DLu, 2017-12-11:
@sradmard This seems to be more of a configuration issue than it is with the code itself. Please make a post on answers.ros.org with the information above. Thanks!

closed 2017-12-11 by DLu

sradmard, 2018-01-03:
@DLu as I am trying to solve my problem, I realised that move_base node is not properly publishing the costmap topics. For example, if I try to check the publish rate of /move_base/local_costmap/costmap, using:
$ rostopic hz /move_base/local_costmap/costmap 
or 
$ rostopic echo /move_base/local_costmap/costmap
you rarely see any information being published (specifically when the robot is stationary). However, if you visualise that topic in rviz, you do see new information on that topic. Please check this video:
[rostopic echo does not publish costmap](https://youtu.be/gvx-dNJ-NVY)

I believe it is because the "seq" field under the "header" is always returned as 0 on the /move_base/local_costmap/costmap topic. I really need to solve this to be able to check the true rate of my costmap being published.

DLu, 2018-01-03:
@sradmard Please debug this on answers.ros.org, not here. Thanks.