ros-planning/navigation #650: AMCL dynamic reconfigure fails

maracuya-robotics, 2018-01-23:
Whenever I try to change a parameter via dynamic reconfigure the parameter is not applied and just a warning is shown in the console:

```
[ WARN] [1516715788.937933420, 40.440000000] [ros.amcl]: Reconfigure callback failed with exception Duration is out of dual 32-bit range:
```

ROS Kinetic, AMCL 1.14.2, running in Gazebo using sim_time.

maracuya-robotics, 2018-01-23:
The cause of this issue is here: https://github.com/ros-planning/navigation/blob/d2fd4f3c73757dfb2f536a6a3755c71c9fb72361/amcl/src/amcl_node.cpp#L485-L486

Especially `save_pose_rate` will be set to `0` if the initial value was `-1` because it is limited by `0` in the dynamic reconfigure configuration.

mikeferguson, 2018-03-25:
Seems straight forward to add some checks that if config.X == 0, print a ROS_ERROR and use the default value from the .cfg file. Marking this as a good first issue for someone who wants to contribute.