ros-planning/navigation #657: global path planner isn't update in indigo (1.12.14)

h-wata, 2018-02-10:
### environment
- ubuntu 14.04
- indigo
- version: 1.12.14(Indigo-devel)
- global_path_planner: navfn
- local_path_planner: base_local_planner
- planner_frequency: 0.5

### Issue
When navigation package is upgraded to 1.12.14 from 1.12.13, global path planner don't update in planner frequency.

DLu, 2018-02-14:
Please open a question on answers.ros.org so that we can have an easier back-and-forth to debug the problem.

closed 2018-02-14 by DLu

pirobot, 2018-03-03:
Exact same problem here.  Even with planner_frequency set > 0, neither NavfnROS nor GlobalPlanner replans when obstacle is moved into path of robot.  Problem appeared immediately after update to 1.12.14.

Compiled indigo-devel branch from source, got same problem.

Compiled kinetic-devel branch from source (compiles fine under Ubuntu 14.04) and problem went away.

So it seems something got changed in Indigo between 1.12.13 and 1.12.14 that broke the planner_frequency functionality.

DLu, 2018-03-20:
Opened #684 for this!

mikeferguson, 2018-03-20:
Fixed in 1.12.15 -- https://github.com/ros/rosdistro/pull/17191