ros-planning/navigation #661: costmap2d does not respect yaw in map origin parameter.

positron96, 2018-02-15:
Hello. I am trying to start a move_base with global costmap loaded from map_server node. The map has origin rotated, the file map.yaml is as follows:
```
image: file.png
resolution: 0.0333
origin: [0, 0, 1]
occupied_thresh: 0.6
free_thresh: 0.2
negate: 0
```

Relevant parts of global costmap config are the following:
```
global_costmap:
  global_frame: /map
  robot_base_frame: base_link
  update_frequency: 4.0
  publish_frequency: 2
  static_map: true

  cost_scaling_factor: 12
  
  
  width: 40.0
  height: 40.0
  resolution: 0.20
  origin_x: 0.0
  origin_y: 0.0

  map_topic: /map
```

When I start, /map topic looks as it should (rotated) but global costmap is not rotated at all.

[image]

Is it the supposed behavior? Can I fix my config to respect rotation?

mikeferguson, 2018-02-15:
Known issue: https://github.com/ros-planning/navigation/issues/166, see also that AMCL does not support this either: https://github.com/ros-planning/navigation/issues/217

Pull requests welcome.

closed 2018-02-15 by mikeferguson

Andrea485, 2020-01-28:
Will this bug ever be fixed?

mikeferguson, 2020-01-29:
Given the remaining lifespan of ROS1, and that ROS2 has it's own, entirely different navigation stack, probably not going to add this feature.

SteveMacenski, 2020-01-29:
If someone made a patch, I'd merge it in ROS2 Navigation, but I'm probably not going to do it myself (ever, frankly).

marcusvinicius178, 2024-04-30:
Hi, I am not being able to effectively move the costmap origin when setting the rolling window mechanism instead of using a static map. Is it not possible? Setting **origin_x and origin_y** in the configuration file is not working. I have described the issue here: https://robotics.stackexchange.com/questions/110892/adjusting-the-origin-of-rolling-costmaps-in-ros2-nav2-for-forward-offset