ros-planning/navigation #689: Trajectory scoring inconsistency

pavloblindnology, 2018-03-27:
According to [this](http://wiki.ros.org/base_local_planner?distro=lunar#Trajectory_Scoring_Parameters) and [this](http://wiki.ros.org/dwa_local_planner?distro=lunar#Trajectory_Scoring_Parameters), the trajectory obstacle scoring is computed as
`occdist_scale * (maximum obstacle cost along the trajectory in obstacle cost (0-254))`
While in the code, `occdist_scale` is multiplied by grid resolution:

- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/base_local_planner/src/trajectory_planner.cpp#L89
- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/dwa_local_planner/src/dwa_planner.cpp#L75

While meter scoring is perfectly appropriate for goal/path distance scoring - imho, it looks very strange and inappropriate for obstacle scoring to depend on map resolution. Bcz, I want to match, say, obstacle cost value 100 to 1 meter distance to path. For this I need `0.01(occdist_scale)*100(cost)` and `1(path_distance_bias)*1(dist_to_path_in_meters)`. This looks naturally and easy to understand. Also, default path and goal biases are set unrealistically high (32.0 and 24.0, respectively). With such values you get final scores like:
`obstacle score = 0.05(resolution) * 0.01(occdist_scale) * 100(cost) = 0.05`
`path score = 32.0(path_distance_bias) * 1(dist_to_path_in_meters) * 0.5 = 16.0`
`goal score = 24.0(goal_distance_bias) * 1(dist_to_goal_in_meters) * 0.5 = 12.0`
which makes obstacle score simply negligible!

In `dwa_local_planner`, additionally, there is inconsistency between next lines
`total_cost = pdist_scale_ * resolution * path_cost + gdist_scale_ * resolution * goal_cost + occdist_scale_ * occ_cost;`
- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/dwa_local_planner/src/dwa_planner.cpp#L200
and 
- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/dwa_local_planner/src/dwa_planner.cpp#L67
- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/dwa_local_planner/src/dwa_planner.cpp#L71
- https://github.com/ros-planning/navigation/blob/df82021fabc6ebdf0d01b86f3082e33dd186f6b9/dwa_local_planner/src/dwa_planner.cpp#L75

As we can see, when computing point cloud scores, path and goal scores aren't additionally multiplied by 0.5 and obstacle score isn't multiplied by resolution at all.

pavloblindnology, 2018-03-30:
Agree about the drastic changes.
May we leave this PR here so that people know about it and update their codes if they wish to?
Or maybe it's better to make some optional release so that the new users of this repo don't follow the same erroneous steps?

DLu, 2018-04-05:
I tend to favor writing new backwards compatible parameters rather than changing them wholesale.

PR #763 (2018-07-18): Local Planner Fixes
 * replaces PR #698 #699
 * fixes #689
 * fixes #697
closed 2018-07-18 by mikeferguson