ros-planning/navigation #701: Drop robot_pose_ekf in Melodic?

mikeferguson, 2018-03-29:
I'm proposing we should drop robot_pose_ekf in Melodic. If you use this package, and intend to in Melodic or later, please comment here.

Reasoning:
 * robot_localization/graft are more generic, and robot_localization is far better documented.
 * robot_pose_ekf has always been fairly PR2-specific (it has several known hacks that aren't generic).
 * PR2 is not going to be supported on Melodic or later AFAIK.

None of the current maintainers of navigation use robot_pose_ekf or know much about the code. Same goes for the BFL dependency. If we are going to continue maintaining this package in Melodic in later, I think we need one of the users to step up and help maintain the package, as well as BFL.

SteveMacenski, 2018-03-30:
While I don't use it anymore, it may be good to have a wiki page for good defaults for the turtlebot series for those users that dont want to get over their head in understanding what those covariance matrices do with `robot_localization`.

mikeferguson, 2018-05-22:
Is in PR -- closing so I can keep track of what we have to actually still do before release!

closed 2018-05-22 by mikeferguson

mkhansen-intel, 2018-07-17:
@mikeferguson - just saw this and thought you might want to also update the wiki. It shows robot_pose_ekf being maintained in melodic. http://wiki.ros.org/robot_pose_ekf

mikeferguson, 2018-07-17:
Interesting -- I'm not sure how we do that since it is automatically generated. @tfoote -- any thoughts on why that shows up that way?

mikeferguson, 2018-07-17:
Oh -- we haven't actually released into Melodic yet -- I bet when we do that, it will get updated!

tfoote, 2018-07-25:
There was at some point a version documented probably from the old branch back in May. 

```
[ros@ros.osuosl.org api]$ ls -lhrt robot_pose_ekf/ navigation/
robot_pose_ekf/:
total 16K
-rw-r--r--. 1 rosbot rosbot    0 May 25 10:16 stamp
-rw-r--r--. 1 rosbot rosbot 1.4K May 25 10:16 manifest.yaml
drwxr-xr-x. 3 rosbot rosbot 8.0K May 25 10:16 html

navigation/:
total 8.0K
-rw-r--r--. 1 rosbot rosbot 1.5K Jul 22 10:10 manifest.yaml
-rw-r--r--. 1 rosbot rosbot    0 Jul 22 10:10 stamp
drwxr-xr-x. 2 rosbot rosbot 4.0K Jul 22 10:10 html
```

We don't delete data on packages that dissappear. So the old data persists. I've manually cleared it.

k-okada, 2019-02-11:
@mikeferguson  @tfoote 
We have maintained bfl (https://github.com/ros-gbp/bfl-release) and PR2 stack (https://github.com/ros-gbp/bfl-release) so very interested in continue maintaining `robot_pose_ekf`. 
Can you create 
https://github.com/ros-planning/robot_pose_ekf and 
https://github.com/ros-gbp/robot_pose_ekf-release ???

Thanks in advance

lukewebdev, 2019-03-20:
I read on the https://github.com/turtlebot/turtlebot/issues/272 issue that robot_pose_ekf is used for create-based turtlebots (I am using a Create 2 on Kinetic, but latest boards I have bought for it only run Ubuntu 18.04.)

k-okada, 2019-04-03:
@davetcoleman @dirk-thomas can we create 
https://github.com/ros-planning/robot_pose_ekf and https://github.com/ros-gbp/robot_pose_ekf-release , so that we can release `robot_pose_ekf` on melodic? I’ll be responsible for these repositories.

mikeferguson, 2019-04-03:
@k-okada repos should now be setup -- along with access for you

k-okada, 2019-04-04:
FYI released into melodic -> https://github.com/ros/rosdistro/pull/20794