ros-planning/navigation #706: RotateRecovery parameter names differ from TrajectoryPlannerROS parameter names

MarcoStb1993, 2018-04-13:
The parameters specified in TrajectoryPlannerROS that according to the documentation will be accepted by RotateRecovery have different than those of RotateRecovery which leads to RotateRecovery using the default values instead of the ones specified by the user in base_local_planners.yaml.
http://wiki.ros.org/rotate_recovery
(acc_lim_th, max_rotational_vel, min_in_place_rotational_vel )
http://wiki.ros.org/base_local_planner?distro=kinetic
(acc_lim_theta, max_vel_theta, min_in_place_vel_theta)

mikeferguson, 2018-05-14:
Adding melodic milestone to this -- since I'd like to try to address all of these random parameter issues we've found over the years in this new LTS.

mikeferguson, 2018-05-22:
Is in PR -- closing so I can keep track of what we have to actually still do before release!

commit d97af143 (2018-05-22): fix param names of RotateRecovery, closes #706
commit 711cfe45 (2018-05-22): fix param names of RotateRecovery, closes #706
commit 01a20d38 (2018-05-29): fix param names of RotateRecovery, closes #706
closed 2018-05-22 by mikeferguson