ros-planning/navigation #710: Thread safety in costmap_2d costmap reset service

SteveMacenski, 2018-04-21:
My colleague and I found the clear costmaps service does not lock the costmap mutex as `updateBounds` and the `clear_costmap_recovery` do. Would adding this lock be something that would be merged?

We find the robot is able to plan in some states when this is called frequently, it will generate a straight line path through the map because there's no costs represented in that iteration.

mikeferguson, 2018-04-21:
Yes, threading fixes are always welcome -- we will probably allow an extended "open comment" period on the pull request so people can (hopefully) test on a variety of platforms before we merge.

SteveMacenski, 2018-04-23:
The changes are in PR #711, its just a couple locks in the service call so I'd doubt anyone would adamantly object. 

I'll check back up on this in a week or two to check in.

SteveMacenski, 2018-05-04:
The PR has been sitting for about 2 weeks without comment. Anything else blocking this I can help with?

mikeferguson, 2018-05-14:
Merged in kinetic -- Lunar forward port will get merged once CI completes (https://github.com/ros-planning/navigation/pull/719)

closed 2018-05-14 by mikeferguson