ros-planning/navigation #712: How do I increase the rate of Robot Pose EKF

RajathSwaroop, 2018-04-23:
I have been using robo pose ekf package from ROS for my project with visual odometry and IMU data being used by ekf package. I am publishing them from different sources and at 20HZ. I get the result topic /robot_pose_ekf/odom_combined at 1.4 hz approximately. I want to increase the rate of output. How do I do that??

SteveMacenski, 2018-04-23:
You'd have to provide your configuration, but if you have that problem I suspect your rate parameter is an issue

`<param name="freq" value="30.0"/>` from wiki.ros.org/robot_pose_ekf

RajathSwaroop, 2018-04-23:
I increased the frequency but could'nt find any change in rate. I tried changing in the launch file as well as the cpp node for robot pose ekf. None of them worked. 

By configuration do you mean setting the flags true or false for the sensor being used??
I have already set in the launch file bool false for odom and true for IMU and Visual Odometry

mikeferguson, 2018-05-14:
This is more appropriate for ROS answers. Please also note, it is our recommendation to use robot_localization on any new robot design -- robot_pose_ekf is slated for removal in melodic.

closed 2018-05-14 by mikeferguson