ros-planning/navigation #714: Navigation on Multiple Robots Conflicting

DanielArnett, 2018-04-28:
I'm trying to use ROS Navigation stack on two robots connected to the same roscore. But both are publishing to `/cmd_vel`, when I need them to publish to `/robot1/cmd_vel` and `/robot2/cmd_vel`. I believe the same is happening with all of my topics related to `move_base`. Every other ROS node I use automatically redirects when I launch it within a `<group ns="robot1">` group tag, but move_base doesn't seem to. Please tell me if I'm not making myself clear. For reference here is the gist of my launch file.

```xml
<launch>
   <group ns="robot1">
  <!--************** Joystick *************** -->
      <arg name="joy_config" default="dualshock4" />
      <arg name="joy_dev" default="/dev/input/js0" />
      <arg name="config_filepath" default="$(find robot)/joystick_config/$(arg joy_config).config.yaml" />

      <node pkg="joy" type="joy_node" name="joy_node">
                <param name="dev" value="$(arg joy_dev)" />
                <param name="deadzone" value="0.3" />
                <param name="autorepeat_rate" value="20" />
      </node>

      <node pkg="teleop_twist_joy" name="teleop_twist_joy" type="teleop_node">
                 <rosparam command="load" file="$(arg config_filepath)" />
      </node>

        <!--  ************** Navigation ***************  -->
	<node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
  	  	<rosparam file="$(find teb_local_planner_tutorials)/cfg/carlike/costmap_common_params.yaml" command="load" ns="global_costmap" />
  	 	<rosparam file="$(find teb_local_planner_tutorials)/cfg/carlike/costmap_common_params.yaml" command="load" ns="local_costmap" />
  		<rosparam file="$(find teb_local_planner_tutorials)/cfg/carlike/local_costmap_params.yaml" command="load" />
  		<rosparam file="$(find teb_local_planner_tutorials)/cfg/carlike/global_costmap_params.yaml" command="load" />
  		<rosparam file="$(find teb_local_planner_tutorials)/cfg/carlike/teb_local_planner_params.yaml" command="load" />

		<param name="base_global_planner" value="global_planner/GlobalPlanner" />
		<param name="planner_frequency" value="1.0" />
		<param name="planner_patience" value="5.0" />

		<param name="base_local_planner" value="teb_local_planner/TebLocalPlannerROS" />
		<param name="controller_frequency" value="5.0" />
		<param name="controller_patience" value="15.0" />

                <param name="clearing_rotation_allowed" value="false" /> <!-- Our carlike robot is not able to rotate in place -->
	</node>
   </group>
</launch>
```

But when I `rostopic list` I get topics like:

```
/move_base/global_costmap/costmap
/move_base/global_costmap/costmap_updates
/move_base/local_costmap/costmap
/move_base/local_costmap/costmap_updates
/move_base/local_costmap/footprint
/move_base_simple/goal
/robot1/joy
/robot1/cmd_vel
```

Notice that all of my other nodes begin with the proper prefix `/robot1`, but move_base does not, even though they're in the same <group> block.

Can anyone think of a good remedy to this?

Thank you.

DanielArnett, 2018-04-28:
The brute force answer was to abuse the remap function.

```xml

  <remap from="/move_base_simple/goal"                                           to="/robot1/move_base_simple/goal" />
  <remap from="/move_base/TebLocalPlannerROS/global_plan"                        to="/robot1/move_base/TebLocalPlannerROS/global_plan" />
  <remap from="/move_base/TebLocalPlannerROS/local_plan"                         to="/robot1/move_base/TebLocalPlannerROS/local_plan" />
  <remap from="/move_base/TebLocalPlannerROS/teb_markers"                        to="/robot1/move_base/TebLocalPlannerROS/teb_markers" />
  <remap from="/move_base/TebLocalPlannerROS/teb_markers_array"                  to="/robot1/move_base/TebLocalPlannerROS/teb_markers_array" />
  <remap from="/move_base/TebLocalPlannerROS/teb_poses"                          to="/robot1/move_base/TebLocalPlannerROS/teb_poses" />
  <remap from="/move_base/global_costmap/costmap"                                to="/robot1/move_base/global_costmap/costmap" />
  <remap from="/move_base/global_costmap/costmap_updates"                        to="/robot1/move_base/global_costmap/costmap_updates" />
  <remap from="/move_base/local_costmap/costmap"                                 to="/robot1/move_base/local_costmap/costmap" />
  <remap from="/move_base/local_costmap/costmap_updates"                         to="/robot1/move_base/local_costmap/costmap_updates" />
  <remap from="/move_base/local_costmap/footprint"                               to="/robot1/move_base/local_costmap/footprint" />
  <remap from="/cmd_vel"                                                         to="/robot1/cmd_vel" />
  <remap from="/move_base/GlobalPlanner/parameter_descriptions"                  to="/robot1/move_base/GlobalPlanner/parameter_descriptions" />
  <remap from="/move_base/GlobalPlanner/parameter_updates"                       to="/robot1/move_base/GlobalPlanner/parameter_updates" />
  <remap from="/move_base/GlobalPlanner/plan"                                    to="/robot1/move_base/GlobalPlanner/plan" />
  <remap from="/move_base/GlobalPlanner/potential"                               to="/robot1/move_base/GlobalPlanner/potential" />
  <remap from="/move_base/TebLocalPlannerROS/obstacles"                          to="/robot1/move_base/TebLocalPlannerROS/obstacles" />
  <remap from="/move_base/TebLocalPlannerROS/parameter_descriptions"             to="/robot1/move_base/TebLocalPlannerROS/parameter_descriptions" />
  <remap from="/move_base/TebLocalPlannerROS/parameter_updates"                  to="/robot1/move_base/TebLocalPlannerROS/parameter_updates" />
  <remap from="/move_base/cancel"                                                to="/robot1/move_base/cancel" />
  <remap from="/move_base/current_goal"                                          to="/robot1/move_base/current_goal" />
  <remap from="/move_base/feedback"                                              to="/robot1/move_base/feedback" />
  <remap from="/move_base/global_costmap/footprint"                              to="/robot1/move_base/global_costmap/footprint" />
  <remap from="/move_base/global_costmap/inflation_layer/parameter_descriptions" to="/robot1/move_base/global_costmap/inflation_layer/parameter_descriptions" />
  <remap from="/move_base/global_costmap/inflation_layer/parameter_updates"      to="/robot1/move_base/global_costmap/inflation_layer/parameter_updates" />
  <remap from="/move_base/global_costmap/obstacle_layer/clearing_endpoints"      to="/robot1/move_base/global_costmap/obstacle_layer/clearing_endpoints" />
  <remap from="/move_base/global_costmap/obstacle_layer/parameter_descriptions"  to="/robot1/move_base/global_costmap/obstacle_layer/parameter_descriptions" />
  <remap from="/move_base/global_costmap/obstacle_layer/parameter_updates"       to="/robot1/move_base/global_costmap/obstacle_layer/parameter_updates" />
  <remap from="/move_base/global_costmap/parameter_descriptions"                 to="/robot1/move_base/global_costmap/parameter_descriptions" />
  <remap from="/move_base/global_costmap/parameter_updates"                      to="/robot1/move_base/global_costmap/parameter_updates" />
  <remap from="/move_base/global_costmap/static_layer/parameter_descriptions"    to="/robot1/move_base/global_costmap/static_layer/parameter_descriptions" />
  <remap from="/move_base/global_costmap/static_layer/parameter_updates"         to="/robot1/move_base/global_costmap/static_layer/parameter_updates" />
  <remap from="/move_base/goal"                                                  to="/robot1/move_base/goal" />
  <remap from="/move_base/local_costmap/obstacle_layer/parameter_descriptions"   to="/robot1/move_base/local_costmap/obstacle_layer/parameter_descriptions" />
  <remap from="/move_base/local_costmap/obstacle_layer/parameter_updates"        to="/robot1/move_base/local_costmap/obstacle_layer/parameter_updates" />
  <remap from="/move_base/local_costmap/parameter_descriptions"                  to="/robot1/move_base/local_costmap/parameter_descriptions" />
  <remap from="/move_base/local_costmap/parameter_updates"                       to="/robot1/move_base/local_costmap/parameter_updates" />
  <remap from="/move_base/local_costmap/static_layer/parameter_descriptions"     to="/robot1/move_base/local_costmap/static_layer/parameter_descriptions" />
  <remap from="/move_base/local_costmap/static_layer/parameter_updates"          to="/robot1/move_base/local_costmap/static_layer/parameter_updates" />
[... cut ...]
  <remap from="/move_base/parameter_updates"                                     to="/robot1/move_base/parameter_updates" />
  <remap from="/move_base/result"                                                to="/robot1/move_base/result" />
  <remap from="/move_base/status"                                                to="/robot1/move_base/status" />
  <remap from="/move_base_simple/goal"                                           to="/robot1/move_base_simple/goal" />
```

Can anyone think of a more elegant solution?

DLu, 2018-04-30:
So I tried a portion of your launch file:

```
<launch>
   <group ns="robot1">
     <!--  ************** Navigation ***************  -->
	   <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
	   </node>
   </group>
</launch>
```
and with that I get 
```
rostopic list
/robot1/cmd_vel
/robot1/move_base/current_goal
/robot1/move_base/goal
/robot1/move_base_simple/goal
/rosout
/rosout_agg
/tf
/tf_static
```
Does that match what you see?

DanielArnett, 2018-05-14:
> Closing due to inactivity and since @DLu <https://github.com/DLu> seems
> to think this is working. If you have additional information, please
> comment/reopen

Yeah it's working.

Before I saw his response working I just remapped every topic related to
move_base and Ros navigation to `/robot1/...`, and that did the trick.

benjaminabruzzo, 2018-11-05:
Adding the group tag does not affect the /move_base/*_costmap topics.

kurshakuz, 2019-08-06:
@benjaminabruzzo did you solve it by using additional remaps with groups or remaps only?