ros-planning/navigation #715: Some local planners don't work with sparse paths

Rayman, 2018-05-09:
The `carrot_planner` outputs a sparse path: just two points. A lot of the local planner (dwa for example) can't deal with this properly. When the starting point falls out of the costmap, the robot stops (this is very easy to reproduce).

What is the correct way to solve this?
1. let the carrot planner and others send out interpolated paths with a lot of intermediate points
2. fix the local planners such that they can deal with sparse paths

I have a fix ready for the carrot planner, but I don't know if that's the right solution.

mikeferguson, 2018-05-09:
The correct way to fix this is to probably not use the carrot_planner. The [carrot_planner wiki](http://wiki.ros.org/carrot_planner) describes how it works, it's really not a useful planner when in any sort of clutter. My understanding is that this "planner" was created for the PR2 beta workshop (the "PR2 with flags demo") -- so they could send robots on straight lines between pre-defined goals with no obstacles. It's also likely that we will drop carrot_planner when we port to ROS2.

Rayman, 2018-05-11:
I'll try to illustrate my problem with a drawing:
[image]
The robot stops immediately when the start point falls out of the local costmap. This is because `getLocalPlan` can't find any point on the path that is inside of the costmap (there are only 2 points).

What is the way to solve this?

mikeferguson, 2018-05-14:
The fix is either to use a different global planner (as I mentioned above) or increase the size of your costmap. Adding interpolation to local planners is likely going to introduce many new parameters into an already complex system. If you really wanted to, you could add interpolation into the carrot planner, but as I mentioned, that's really not intended to be a real planner.

I suppose we would merge a PR that adds interpolated points in the carrot planner, but I would not consider it a priority or concern for most users, and as such, I'm going to close this issue.

closed 2018-05-14 by mikeferguson