ros-planning/navigation #741: Tuning move_base for a large robot

shahvi, 2018-06-11:
Hi, I am trying to implement the move_base on the Clearpath warthog platform which is about 1.8x1.6mts in size.  I am having a couple of issues and was hoping someone could help me address them:
1. When directing the robot to navigate to a point, I find that once it reaches the x,y coodrinates, the planner makes it go almost full circle to get to the desired orientation.  
2. I find that this large platform doesn't respond to small values of cmd_velocities in many cases it needs as large as 0.4rads/sec before the feels actually rotate in place.
In order to deal with 2, I have created a velocity amplifier node, which amplifies the cmd_velocities by a factor of 10 to help it actually move the wheels.  Is there a better way to deal with this in move_base?

Any thoughts on how to avoid the in-place complete rotation descried in 1?

Here are the parameters I am using:
```base_global_planner: navfn/NavfnROS
base_local_planner: base_local_planner/TrajectoryPlannerROS
recovery_behaviors: 
  - {name: conservative_reset, type: clear_costmap_recovery/ClearCostmapRecovery}
  - {name: aggressive_reset, type: clear_costmap_recovery/ClearCostmapRecovery}
#recovery_behaviors: [{name: rotate_recovery, type: rotate_recovery/RotateRecovery}]
controller_frequency: 20.0
planner_patience: 5.0
controller_patience: 15.0
conservative_reset_dist: 3.0
recovery_behavior_enabled: false
#clearing_rotation_allowed: false
shutdown_costmaps: false
aggressive_reset: {reset_distance: 1.84}
oscillation_timeout: 30.0
oscillation_distance: 0.5
planner_frequency: 0.0
max_planning_retries: -1
footprint: [[-0.76, -0.7], [-0.76, 0.7], [0.8, 0.7], [0.8, -0.7]]
footprint_padding: 0.01

global_costmap:
  global_frame: /map
  robot_base_frame: /base_link
  transform_tolerance: 0.2
  update_frequency: 5.0
  publish_frequency: 2.0
  rolling_window: false
  always_send_full_costmap: false
  width: 40
  height: 40
  origin_x: 0.0
  origin_y: 0.0
  resolution: 0.1
  plugins:
  - {name: static_layer, type: 'costmap_2d::StaticLayer'}
  - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
  - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
  
  static_layer:
    map_topic: map
    first_map_only: false
    track_unknown_space: true
    trinary_costmap: true
#    static_map: false

  obstacle_layer:
    track_unknown_space: true  
    footprint_clearing_enabled: true
    max_obstacle_height: 2.0
    obstacle_range: 10.0
    raytrace_range: 5.0
    observation_sources: scan_filtered
    scan_filtered: 
      topic: /scan_filtered
      observation_persistence: 0.0
      expected_update_rate: 0.1 
      data_type: LaserScan
      clearing: true
      marking: false
      inf_is_valid: false
#    combination_method: 1   
  inflation_layer:
    inflation_radius: 0.01
    cost_scaling_factor: 10.0
    
local_costmap:
  global_frame: /odom
  robot_base_frame: /base_link
  transform_tolerance: 0.2
  update_frequency: 10.0
  publish_frequency: 1.0
  rolling_window: true
  always_send_full_costmap: false
  width: 8
  height: 8
  origin_x: 0.0
  origin_y: 0.0
  resolution: 0.05
  plugins:
  - {name: obstacle_layer, type: 'costmap_2d::ObstacleLayer'}
  - {name: inflation_layer, type: 'costmap_2d::InflationLayer'}
  
  obstacle_layer:
    track_unknown_space: true  
    footprint_clearing_enabled: true
    max_obstacle_height: 2.0
    obstacle_range: 10.0
    raytrace_range: 5.0
    observation_sources: scan_filtered
    scan_filtered: 
      topic: /scan_filtered
      observation_persistence: 0.0
      expected_update_rate: 0.1 
      data_type: LaserScan
      clearing: true
      marking: true
      inf_is_valid: false
#    combination_method: 1   
  inflation_layer:
    inflation_radius: 0.01
    cost_scaling_factor: 10.0
    
NavfnROS:
  allow_unknown: false
  planner_window_x: 0.0
  planner_window_y: 0.0
  default_tolerance: 0.2
  visualize_potential: false

TrajectoryPlannerROS:
  max_vel_x: 1.0
  min_vel_x: 0.1
  max_vel_theta: 1.0
  min_vel_theta: -1.0
  min_in_place_vel_theta: 0.0
  escape_vel: -0.10
  acc_lim_x: 1.0
  acc_lim_y: 1.0
  acc_lim_theta: 5.0

  holonomic_robot: false
  yaw_goal_tolerance: 0.25 # about 15 degrees
  xy_goal_tolerance: 0.25 # 10 cm
  latch_xy_goal_tolerance: true
  
  sim_time: 10.0
  sim_granularity: 0.1
  angular_sim_granularity: 0.025
  vx_samples: 8
  vtheta_samples: 20
  
  meter_scoring: true
  pdist_scale: 1.2 # def 0.6
  gdist_scale: 0.8
  occdist_scale: 0.05
  heading_lookahead: 1.0
  heading_scoring: false
  heading_scoring_timestep: 0.8
  dwa: true
  publish_cost_grid_pc: true
  global_frame_id: odom
  oscillation_reset_dist: 0.1
  prune_plan: true
# simple_attractor: false```

mikeferguson, 2018-06-11:
@shahvi -- please post future questions to ROS answers. We try to limit issues to actual "issues" within the navigation stack code base.

RE: rotation direction issues. I know that existed as an issue long ago -- I thought it was fixed. See https://github.com/ros-planning/navigation/pull/348

RE: slow rotation -- see the parameter min_in_place_vel_theta. Amplifying the output is certain to cause problems (the planner can't accurately predict what is going to happen).

closed 2018-06-11 by mikeferguson

shahvi, 2018-06-11:
Understood... thanks for the info here.

shahvi, 2018-06-12:
@mikeferguson  After understanding the issue a bit more, I think the planner can benefit from an additional parameter for minimum velocity theta (while still moving).  This would help the planner model the robot's capabilities better. 
I understand that there is a parameter for minimum velocity theta already, however that parameter defaults to -1.0.  If I set it to a positive number such as 0.4, I see that it only considers velocities between 0.4 and 1.0 which is not desirable.
I suggest flagging this as an enhancement, unless I misunderstand something.

darshb34, 2020-10-14:
Hi @shahvi , I am facing the same issue where the robot keep rotating in place once it reaches it's goal. It gives me error "the controller is oscillating for 5.0568s". Any suggestions please?

Thank you

shahvi, 2020-10-14:
First this to try would be to reduce both `max_vel_theta` and `min_vel_theta` and / or reducing `acc_lim_theta` which should prevent the robot from overshooting. You can also try increasing the `yaw_goal_tolerance` which let it know that it has reached the destination even if yaw is not perfect.
Depending on how accurate you need the yaw to be, one practical thing that can help is to set an initial goal couple meters away from the intended goal and then do the final approach as a separate command. Hope this helps.