ros-planning/navigation #744: New release for Kinetic

alireza-hosseini, 2018-06-19:
Hi,

We are using google's Cartographer in combination with move-base for an exploration task and move-base is constantly crashing while resizing the map with the same error logs reported on #584. 
Apprantly this issue is fixed on #675, however, the latest release for **Kinetic** (956c2c9e6fddce5484343cfc97dbc4e7217edcc2) is triggered exactly one week before merging the mentioned PR. So It would be really great if we have a new release since it is blocking us and probably many others.

alireza-hosseini, 2018-06-20:
Thanks for your quick response.

wennycooper, 2018-11-21:
Sorry that this is not related to this issue, but I'm quite confusing about how the release been done.
Would you please tell me:
1. How to get the list of previous release notes of navigation?
2. How the release been done?
Say, Is there anything automatic to collect packages and build all of them to be a "release"? 
Where and when and who to make it happen?

mikeferguson, 2018-11-21:
@wennycooper 

Each package has a CHANGELOG.rst that details what was new in that release (and also the release dates).

Cutting a "release" means we (the maintainers) run catkin_prepare_release and bloom -- which makes a new version available to install via debians. After a release, it is often 1-3 weeks before OSRF syncs those new packages to be publicly available.