ros-planning/navigation #747: How to install the latest navigation

ytgcljj, 2018-06-22:
I installed ROS with the command of sudo apt-get install ros-kinetic-desktop-full. Now I'm updating navigation. I don't know how to update it. Please help me

mikeferguson, 2018-06-28:
This is really a question for answers.ros.org.

closed 2018-06-28 by mikeferguson

mikeferguson, 2018-08-14:
The ROS wiki covers how to install packages in general -- http://wiki.ros.org/melodic/Installation/Ubuntu -- see the section on "installing an individual package". The ROS wiki also does this for ALL available platforms -- not just Ubuntu.

If you're trying to jump straight into navigation without learning how to use ROS in general, you're doing it wrong. The documentation of every individual package shouldn't include the full set of instructions for how to install on every available platform.