ros-planning/navigation #749: Melodic Release?

bochen87, 2018-06-25:


mikeferguson, 2018-06-25:
Yeah, we're a bit behind -- see the milestone for Melodic (https://github.com/ros-planning/navigation/milestone/4) -- I think we may end up pushing those remaining items (I was really hoping to have fully switched to TF2, since so many core stacks have -- but I've not yet been able to actually make it work)

bochen87, 2018-06-26:
Oh, ok, thanks for the explanation! I was just wondering because it wasn't noted anywhere that the navigation stack was not ready yet. In the installation it even mentions that navigation get's installed with `ros-melodic-desktop-full` so I first spend half an hour looking for potential errors on my computer :-)
Unfortunately, I don't have access rights yet, but maybe worth adding this as a disclaimer in the installation notes on the wiki? http://wiki.ros.org/action/login/melodic/Installation/Ubuntu

bochen87, 2018-06-26:
Oh, I see - well maybe also additionally mentioning in the melodic documentation that the navigation is not yet installable as package. I asked for edit rights in the wiki, will have to wait until I get that, then I can update it if you want to.

mikeferguson, 2018-06-26:
Putting temporary notes on the release status on the Wiki is probably a bad idea -- nobody will end up removing it later on. It's better that people look at the automatically generated and always up to date [status pages](http://repositories.ros.org/status_page/ros_melodic_default.html) (which I will admit are somewhat hard to find -- which is something that should be addressed). If something isn't listed (like navigation) then it hasn't been released. Once it is released, that page makes it easy to know if it has been synced to public, or if new releases are in the pipeline.

bochen87, 2018-06-27:
I removed the mention of the navigation stack in the desktop-full package. Can we leave this issue open so that people know the navigation stack is not available as a package on melodic as of yet?

mikeferguson, 2018-06-27:
Yep, I’ve added this ticket to the milestone, and will leave it open

mikeferguson, 2018-07-25:
I'm working on release now

mikeferguson, 2018-07-25:
Release is tagged -- bloom is taking foreveeeeeeeer

mikeferguson, 2018-07-25:
https://github.com/ros/rosdistro/pull/18657

mikeferguson, 2018-07-25:
Will work on the wiki changelog later tonight

closed 2018-07-25 by mikeferguson

subodh-malgonde, 2018-08-14:
After going through this thread, I understand that the navigation package has been released for ROS Melodic. Am I right?

I am on Ubuntu 18.04 and I am still unable to install navigation as a package. Please correct me if I am wrong.

mikeferguson, 2018-08-14:
It has been released, but a sync has not yet happened. We have no control over when OSRF does a sync.