ros-planning/navigation #750: Already moving robot does not stop if base_local_planner can not make a plan

sunilsulania, 2018-06-26:
I have a robot running in a physical loop path with move_base running only in one of its straight segments. So i switch (Give goal to actionserver) to move_base while robot is moving. 
In a condition when **MoveBase::makePlan** return false immediately after switching(giving goal to actionserver) velocities are not updated to zero causing robot to not stop. 
makePlan returns false because there is an obstacle covering goal. And as an effect of this, robot goes crashing into obstacle.

I think I got the reason for it :

```
bool gotPlan = n.ok() && makePlan(temp_goal, *planner_plan_);

if(gotPlan){
.
.
.}
else if(state_==PLANNING)
{
.
.
    if(runPlanner_ && (ros::Time::now() > attempt_end || planning_retries_ > uint32_t(max_planning_retries_)))
    {
        .
        publishZeroVelocity();
        .
    }
.
.}
```
`(ros::Time::now() > attempt_end || planning_retries_ > uint32_t(max_planning_retries_))` This term is false and robot does not stop

mikeferguson, 2018-06-27:
I think I might have seen this before and was blaming the simulation. Could you open a PR?

mikeferguson, 2018-06-28:
Ok -- I think this is different from what I've seen -- but I think you're right -- I don't see anywhere in the executeCb() where we set velocities to zero. That should probably be done right before we change state to planning.

DLu, 2019-06-18:
Closed via #751

closed 2019-06-18 by DLu