ros-planning/navigation #758: Jenkins/Docker setup is slower after tf2 #752

moriarty, 2018-07-16:
[image]

The part which has slowed down seems to be the pre-setup stages. I thought maybe it is because some template code in tf2 slows things down but it doesn't seem so. It now takes **much** loger to get to the build workspace in isolation stage.

Build 31
http://build.ros.org/job/Mpr__navigation__ubuntu_bionic_amd64/31/consoleFull#console-section-15
```
04:21:46 GitHub pull request #751 of commit d84d855ab344f6c1f3c8c84756fa42c326a9283e, no merge conflicts.
...
# BEGIN SUBSECTION: build workspace in isolation
04:36:25 Invoking '. /opt/ros/melodic/setup.sh && PYTHONIOENCODING=utf_8 PYTHONUNBUFFERED=1 catkin_make_isolated --cmake-args -DCATKIN_ENABLE_TESTING=1 -DCATKIN_SKIP_TESTING=0 -DCATKIN_TEST_RESULTS_DIR=/tmp/catkin_workspace/test_results --catkin-make-args -j1' in '/tmp/catkin_workspace'
```

Build 32
http://build.ros.org/job/Mpr__navigation__ubuntu_bionic_amd64/32/consoleFull#console-section-15
```
21:06:52 GitHub pull request #752 of commit 88ada10d9129d529a373162f42fb74cd0821d649, no merge conflicts.
...
# BEGIN SUBSECTION: build workspace in isolation
22:30:01 Invoking '. /opt/ros/melodic/setup.sh && PYTHONIOENCODING=utf_8 PYTHONUNBUFFERED=1 catkin_make_isolated --cmake-args -DCATKIN_ENABLE_TESTING=1 -DCATKIN_SKIP_TESTING=0 -DCATKIN_TEST_RESULTS_DIR=/tmp/catkin_workspace/test_results --catkin-make-args -j1' in '/tmp/catkin_workspace'
```

mikeferguson, 2018-07-16:
The difference in getting to build start is not actually different -- that's just how long it took for a buildslave to be available. For instance, here is a successful build from a while ago, that took 21 minutes to start as well: http://build.ros.org/job/Mpr__navigation__ubuntu_bionic_amd64/7/console

moriarty, 2018-07-16:
Okay, and it actually looks like the restarted build for the #755 PR is already far along and should pass. http://build.ros.org/job/Mpr__navigation__ubuntu_bionic_amd64/38/consoleFull#console-section-12

Then should I close this?

mikeferguson, 2018-07-16:
Leave it open -- we've not had this many issues until right now -- so I think something is wonky -- just not sure what yet

mikeferguson, 2018-07-16:
And actually you are right -- I think I was looking at the wrong timestamps initially. Doing an elapsed view:

 * First 11 steps went from 4 minutes to 36 minutes
 * Takes us 1hr35min to start tests, instead of 22minutes
 * Tests take about 20 minutes in both -- but if they run a little long, we blow through the time limit.

mikeferguson, 2018-07-16:
Build 38 appears to have started tests after 20 minutes -- so maybe this was actually a build farm issue? @tfoote -- any thoughts?

mikeferguson, 2018-07-18:
Seems to have been intermittent on buildfarm and gone away...

closed 2018-07-18 by mikeferguson