ros-planning/navigation #759: CostMap2d dependency missing - catkin_make fails

HobbySingh, 2018-07-16:
make[2]: *** No rule to make target '/usr/lib/libpcl_common.so', needed by '/home/mandeep/catkin_ws/devel/lib/libcostmap_2d.so'.  Stop.
CMakeFiles/Makefile2:17453: recipe for target 'navigation/costmap_2d/CMakeFiles/costmap_2d.dir/all' failed
make[1]: *** [navigation/costmap_2d/CMakeFiles/costmap_2d.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

mikeferguson, 2018-07-16:
cost map_2d is building in CI -- so it is unlikely we are missing dependencies. Please use rosdep to install the dependencies when building from source.

closed 2018-07-16 by mikeferguson

HobbySingh, 2018-07-16:
My sincere apologies. 
The issue was that my libpcl_common was not present in /usr/lib/ but it was there in /usr/lib/x86_64-linux-gnu. I don't know what does that mean. I simply copy psted all the missing libpcl_* files to /usr/lib folder and solved.