ros-planning/navigation #760: PR #752 broke orientation filter

Rayman, 2018-07-16:
Hey @moriarty, after your PR, the orientation filter doesn't work anymore. I don't see anyting obvious wrong in the code, but it behaves differently. See the screenshots below.

melodic-devel (96c5f9fce6bdc0128623b3e4b184bcbe2909403d)
[image]

8ea5462f5395acf9741253ff38302d29175dd870
[image]

moriarty, 2018-07-16:
Damn. I didn’t catch that when I was testing #752, but that might explain something I saw with #755.

I will take a look

moriarty, 2018-07-16:
I think this is a difference in ```tf2::getYaw``` @Rayman can you try replacing that with ```tf2::getEulerYPR``` ?

The old under the hood looks like it called getRPY and threw away the pitch and roll.

http://docs.ros.org/kinetic/api/tf/html/c++/transform__datatypes_8h_source.html#l00138
```
  131 static inline double getYaw(const Quaternion& bt_q){
  132   tfScalar useless_pitch, useless_roll, yaw;
  133   tf::Matrix3x3(bt_q).getRPY( useless_roll, useless_pitch,yaw);
  134   return yaw;
  135 }
  136 
  138 static inline double getYaw(const geometry_msgs::Quaternion& msg_q){
  139   Quaternion bt_q;
  140   quaternionMsgToTF(msg_q, bt_q);
  141   return getYaw(bt_q);
  142 }
```

http://docs.ros.org/kinetic/api/tf/html/c++/Matrix3x3_8h_source.html#l00365
```
  365         void getRPY(tfScalar& roll, tfScalar& pitch, tfScalar& yaw, unsigned int solution_number = 1) const
  366         {
  367         getEulerYPR(yaw, pitch, roll, solution_number);
  368         }
```

http://docs.ros.org/jade/api/tf2/html/impl_2utils_8h_source.html
```
00086 inline
00087 void getEulerYPR(const tf2::Quaternion& q, double &yaw, double &pitch, double &roll)
00088 {
00089   double sqw;
00090   double sqx;
00091   double sqy;
00092   double sqz;
00093 
00094   sqx = q.x() * q.x();
00095   sqy = q.y() * q.y();
00096   sqz = q.z() * q.z();
00097   sqw = q.w() * q.w();
00098 
00099   // Cases derived from https://orbitalstation.wordpress.com/tag/quaternion/
00100   double sarg = -2 * (q.x()*q.z() - q.w()*q.y()) / (sqx + sqy + sqz + sqw); /* normalization added from urdfom_headers */
00101   if (sarg <= -0.99999) {
00102     pitch = -0.5*M_PI;
00103     roll  = 0;
00104     yaw   = -2 * atan2(q.y(), q.x());
00105   } else if (sarg >= 0.99999) {
00106     pitch = 0.5*M_PI;
00107     roll  = 0;
00108     yaw   = 2 * atan2(q.y(), q.x());
00109   } else {
00110     pitch = asin(sarg);
00111     roll  = atan2(2 * (q.y()*q.z() + q.w()*q.x()), sqw - sqx - sqy + sqz);
00112     yaw   = atan2(2 * (q.x()*q.y() + q.w()*q.z()), sqw + sqx - sqy - sqz);
00113   }
00114 };
00115 
00122 inline
00123 double getYaw(const tf2::Quaternion& q)
00124 {
00125   double yaw;
00126 
00127   double sqw;
00128   double sqx;
00129   double sqy;
00130   double sqz;
00131 
00132   sqx = q.x() * q.x();
00133   sqy = q.y() * q.y();
00134   sqz = q.z() * q.z();
00135   sqw = q.w() * q.w();
00136 
00137   // Cases derived from https://orbitalstation.wordpress.com/tag/quaternion/
00138   double sarg = -2 * (q.x()*q.z() - q.w()*q.y()) / (sqx + sqy + sqz + sqw); /* normalization added from urdfom_headers */
00139 
00140   if (sarg <= -0.99999) {
00141     yaw   = -2 * atan2(q.y(), q.x());
00142   } else if (sarg >= 0.99999) {
00143     yaw   = 2 * atan2(q.y(), q.x());
00144   } else {
00145     yaw   = atan2(2 * (q.x()*q.y() + q.w()*q.z()), sqw + sqx - sqy - sqz);
00146   }
00147   return yaw;
00148 };
00149 
00150 }
00151 }
00152 
00153 #endif //TF2_IMPL_UTILS_H
```

mikeferguson, 2018-07-16:
@moriarty -- I added some comments in #752 -- it looks like the calls to setEuler should maybe setRPY if you look at the bodies of the functions in tf and tf2.

closed 2018-07-18 by mikeferguson