ros-planning/navigation #761: kinetic-level cannot finish compile

Kanelankai, 2018-07-16:
My ROS version is kinetic.
So I download the version of kinetic-level to the src folder of my workspace.
However, compile stopped as follows:
"
.....
stmap_2d_markers
[ 75%] Built target costmap_2d_node
[ 76%] Built target layers
[ 76%] Built target point_grid
[ 76%] Built target clear_costmap_recovery
[ 76%] Built target move_slow_and_clear
[ 77%] Built target navfn
[ 91%] Built target darknet_ros_lib
[ 94%] Built target base_local_planner
[ 94%] Built target navfn_node
[ 94%] Built target darknet_ros
[ 95%] Built target global_planner
[ 96%] Built target trajectory_planner_ros
[ 97%] Built target planner
[ 98%] Built target rotate_recovery
[ 98%] Built target carrot_planner
[ 98%] Built target dwa_local_planner
[100%] Built target move_base
[100%] Built target move_base_node

"
It stopped here with no error information.
Could someone tell me what happened when compiling this package?

mikeferguson, 2018-07-16:
If CTRL-C and re-run catkin-make, does it work? I'm guessing this is actually a catkin issue, not something in our code, since it didn't error out...

Kanelankai, 2018-07-17:
Yes. I have tried to re-run catkin_make but nothing changed. However, other packages could finish compile except this one.
I will compile it with other pc to check whether the settings of my pc have some problems.
Thank you!
iPhoneから送信

Kanelankai, 2018-07-17:
I have solved this problem.
I wait for about 15min today and finally it finished!
Although I did the same thing yesterday.
Sorry for disturbing you.
Thank you very much!

Kanelankai, 2018-07-17:
@sunilsulania I first tried to compile this package last night, my pc hung up for a long time with no reaction. I tried many times but with no success. I began to doubt whether something was wrong with ros. Fortunately compilation finally finished when I tried again this morning. It takes too much time to finish move_base_node.
Thank you.

mikeferguson, 2018-07-17:
We haven't changed much in Kinetic lately -- so I'm thinking this might be some upstream issue in catkin_make. You might try looking around about catkin_make having this with other packages -- if you find there is something we can do to fix the build time, please open a new issue (or even better, a pull request) with the changes we should look at.

closed 2018-07-17 by mikeferguson