ros-planning/navigation #764: Need advice about new AMCL feature implementation

AchmadFathoni, 2018-07-18:
I am making a robot for competition that must localize fast for solving kidnapped robot problem in [this map](https://novikaginanto.files.wordpress.com/2012/12/lapangan-berkaki.jpg). For doing that I have three new feature that not in AMCL repository yet and I need advice about how to implement it:

1. Pose elemination using compass heading data
    Says I have very good compass with 5 degress error and I want any particle that has heading that differ    more than 5 degress from compass reading is eliminated, not only at initial pose but as long the particle filter running.
2. Specific initial pose distribution
    My robot always start inside room(see the map) but by default AMCL distributes particles evenly on entire map region. For initial guess I only want to set the particle inside the room with heading that respect feature number 1.
3. Use robot radius as minimum distance from grid map obstacle
    My robot radius is about 12.5 cm but AMCL distributes particle too close to the wall with distance less than robot radius. Implement this feature will make AMCL more accurate.

procopiostein, 2018-07-18:
https://answers.ros.org/questions/ is a better place for those type of questions.
the purpose of this page is for discussing bugs, enhancements, etc...
please, repost your question on answers.ros and close this issue.
thanks

mikeferguson, 2018-07-18:
Thanks @procopiostein -- that's right.

The only thing I will add -- having done robot fire fighting contests before -- depending on the sensor noise, each room should look "fairly different". You might be able to just do a brute force geometry match and call AMCL with an initial pose.

closed 2018-07-18 by mikeferguson