ros-planning/navigation #767: Compilation issue

NalinLuthra, 2018-07-23:
Can't compile AMCL package  

error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
     std::shared_ptr<tf2_ros::TransformBroadcaster> tfb_;
          ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:144:10: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
     std::shared_ptr<tf2_ros::TransformListener> tfl_;
          ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:145:10: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
     std::shared_ptr<tf2_ros::Buffer> tf_;
          ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In constructor ‘AmclNode::AmclNode()’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:433:3: error: ‘tfb_’ was not declared in this scope
   tfb_.reset(new tf2_ros::TransformBroadcaster());
   ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:434:3: error: ‘tf_’ was not declared in this scope
   tf_.reset(new tf2_ros::Buffer());
   ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:435:3: error: ‘tfl_’ was not declared in this scope
   tfl_.reset(new tf2_ros::TransformListener(*tf_));
   ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In member function ‘void AmclNode::reconfigureCB(amcl::AMCLConfig&, uint32_t)’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:614:63: error: ‘tf_’ was not declared in this scope
                                                              *tf_,
                                                               ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In member function ‘void AmclNode::runFromBag(const string&)’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:674:9: error: ‘tf_’ was not declared in this scope
         tf_->setTransform(tf_msg->transforms[ii], "rosbag_authority");
         ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In member function ‘bool AmclNode::getOdomPose(geometry_msgs::PoseStamped&, double&, double&, double&, const ros::Time&, const string&)’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:977:11: error: ‘class AmclNode’ has no member named ‘tf_’
     this->tf_->transform(ident, odom_pose, odom_frame_id_);
           ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In member function ‘void AmclNode::laserReceived(const LaserScanConstPtr&)’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1093:13: error: ‘class AmclNode’ has no member named ‘tf_’
       this->tf_->transform(ident, laser_pose, base_frame_id_);
             ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1215:7: error: ‘tf_’ was not declared in this scope
       tf_->transform(min_q, min_q, base_frame_id_);
       ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1400:15: error: ‘class AmclNode’ has no member named ‘tf_’
         this->tf_->transform(tmp_tf_stamped, odom_to_map, odom_frame_id_);
               ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1423:15: error: ‘class AmclNode’ has no member named ‘tfb_’
         this->tfb_->sendTransform(tmp_tf_stamped);
               ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1445:13: error: ‘class AmclNode’ has no member named ‘tfb_’
       this->tfb_->sendTransform(tmp_tf_stamped);
             ^
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp: In member function ‘void AmclNode::handleInitialPoseMessage(const PoseWithCovarianceStamped&)’:
/home/nalin/Navigation/catkin_ws/src/navigation/amcl/src/amcl_node.cpp:1491:15: error: ‘tf_’ was not declared in this scope
     tx_odom = tf_->lookupTransform(base_frame_id_, msg.header.stamp,

mikeferguson, 2018-07-23:
Which version of Linux/ROS are you using? I presume you have the melodic-devel branch checked out on an older Linux/ROS since our CI is building just fine.

closed 2018-07-23 by mikeferguson

salahkhan94, 2018-07-27:
I am facing the same issue. I am trying to compile it on ROS Kinetic