ros-planning/navigation #768: Global costmap resized by static layer even if layered_costmap_->isSizeLocked() is set to true

jinder1s, 2018-07-25:
In the function staticLayer::incomingMap(navigation/costmap_2d/plugins/static_layer.cpp line 166), the first if statement results in master map(layered_costmap) being resized. This seems unintended to me and I was wondering the changes below should be made. There are few other changes to be made for this to fully work, but for now I just wanted to ask if what is currently happening matches with communities idea of what should be happening.

///FROM this***********************************************************
if (!layered_costmap_->isRolling() && (master->getSizeInCellsX() != size_x ||
      master->getSizeInCellsY() != size_y ||
      master->getResolution() != new_map->info.resolution ||
      master->getOriginX() != new_map->info.origin.position.x ||
      master->getOriginY() != new_map->info.origin.position.y ||
      !layered_costmap_->isSizeLocked()))
  {
    // Update the size of the layered costmap (and all layers, including this one)
    ROS_INFO("Resizing costmap to %d X %d at %f m/pix", size_x, size_y, new_map->info.resolution);
    layered_costmap_->resizeMap(size_x, size_y, new_map->info.resolution, new_map->info.origin.position.x,
                                new_map->info.origin.position.y, true);
}

///TO this***************************************************************
if (!layered_costmap_->isRolling() && !layered_costmap_->isSizeLocked() && (master->getSizeInCellsX() != size_x ||
      master->getSizeInCellsY() != size_y ||
      master->getResolution() != new_map->info.resolution ||
      master->getOriginX() != new_map->info.origin.position.x ||
      master->getOriginY() != new_map->info.origin.position.y))
  {
    // Update the size of the layered costmap (and all layers, including this one)
    ROS_INFO("Resizing costmap to %d X %d at %f m/pix", size_x, size_y, new_map->info.resolution);
    layered_costmap_->resizeMap(size_x, size_y, new_map->info.resolution, new_map->info.origin.position.x,
                                new_map->info.origin.position.y, true);
}

moriarty, 2018-07-26:
Sorry this comment won't answer your question. I just saw a wall of unformatted text and had to say something.

I guess there is no "how to file a bug" for this repository, but some more strict repos would require a little bit more basic information for [example](https://github.com/ev3dev/ev3dev/blame/ev3dev-jessie/ISSUE_TEMPLATE.md)

FYI, if you find the lines in GitHub, you can click on them, and then select "copy permalink" which is much easier to read... Then you can also use three ` marks, to type code blocks and use diff format to show clearer what you mean.

https://github.com/ros-planning/navigation/blob/7fa21a2759806ffea7fee9be436ab27b6b3c102f/costmap_2d/plugins/static_layer.cpp#L174-L185

I _might_ have made a typo, but from reading your text and typing it again (this is easier if you try it on your machine first and just copy some of the output of git diff), I think you want to replace the lines above, with:

```diff
-  if (!layered_costmap_->isRolling() && (master->getSizeInCellsX() != size_x ||
+  if (!layered_costmap_->isRolling() && !layered_costmap_->isSizeLocked()
+   && (master->getSizeInCellsX() != size_x ||
    master->getSizeInCellsY() != size_y ||
    master->getResolution() != new_map->info.resolution ||
    master->getOriginX() != new_map->info.origin.position.x ||
-   master->getOriginY() != new_map->info.origin.position.y ||
-   !layered_costmap_->isSizeLocked()))
+   master->getOriginY() != new_map->info.origin.position.y))
```

A quick look at git-blame view from within GitHub shows that these lines were last modified with PR https://github.com/ros-planning/navigation/pull/324 which was @mikeferguson & @reinzor

jinder1s, 2018-07-26:
Thanks for the suggestions! Will update and reopen with corrections @moriarty

moriarty, 2018-07-27:
@jinder1s I didn’t mean be rude or scare you away, I just wanted to help pin point the issue you were trying to report

jinder1s, 2018-07-27:
@moriarty  no no, I genuinely appreciate the advice. Its just the confluence of many events yesterday did not allow me to edit this issue request till now.

DLu, 2019-06-18:
Closed via #792

closed 2019-06-18 by DLu