ros-planning/navigation #775: cost2d

lvxingzhe1991, 2018-07-31:
Could not find a package configuration file provided by "tf2_sensor_msgs"
  with any of the following names:

    tf2_sensor_msgsConfig.cmake
    tf2_sensor_msgs-config.cmake
Add the installation prefix of "tf2_sensor_msgs" to CMAKE_PREFIX_PATH or
  set "tf2_sensor_msgs_DIR" to a directory containing one of the above files.
  If "tf2_sensor_msgs" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  navigation/costmap_2d/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mtkj/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/mtkj/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:4924: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

mikeferguson, 2018-07-31:
You need to install tf2_sensor_msgs -- it is properly in the package.xml as a build depend.

closed 2018-07-31 by mikeferguson

Oshoriame, 2019-10-18:
Hello, were you able to solve this problem. I tried installing tf2_sensor_msgs and i got this error
E: Unable to locate package ros-kinetic-tf2-sensor-msgs

Napoleon-BlownApart, 2020-02-15:
Is the OP about `kinetic` or `melodic`?  I'm having this issue with `melodic`.

kartik-pujar, 2022-03-17:
same here broo

shrinivas96, 2022-04-03:
If someone is still looking for a solution to this problem:

Go to the official page of [tf2_sensor_msgs](http://wiki.ros.org/tf2_sensor_msgs) and click on your ROS distribution. Down there it lists the git repository along with the branch name that contains the package for your ROS distribution (for me it was noetic-devel). Then download the repository ZIP file, unpack it in a directory.

Then, place the symlink of this extracted folder inside your `~/path/to/your/catkin_ws/src` folder and build your package again. 

I prefer this way to installing packages using apt purely because I do not wish to populate my system with packages that I might need only for some time.