ros-planning/navigation #782: How to reduce the density of the generated global path?

mojovski, 2018-08-29:
I would like to reduce the number of poses published in the move_base/Navfn/plan topic, which is sometimes a few thousands in my case.
Is there a way to configure this with params?
If no, can someone point me to the source code, where I might extend it?

Thank you very much!

DLu, 2018-08-29:
Please post questions like this on answers.ros.org in the future. The easiest way to reduce the number of poses is to reduce the resolution of your costmap. Other than that, I can't think of a direct way to simplify the path.

closed 2018-08-29 by DLu

mikeferguson, 2018-08-29:
For the navfn planner, the publisher is located here: https://github.com/ros-planning/navigation/blob/melodic-devel/navfn/src/navfn_ros.cpp#L343 -- you could I suppose add some downsampling there (although the plan published might no longer be very true to the real plan)

commit 3c9e346f (2019-06-18): Fixes #782 (#892)