ros-planning/navigation #786: Undefined reference in navfn library

ipa-fez, 2018-09-05:
https://github.com/ros-planning/navigation/blob/c7bcd005fde62d3eaa2069b96f33ba4c472eadb8/navfn/include/navfn/navfn.h#L96

The function in the cpp file doesn't use a const COSTMAP* which means that the declaration and definition don't match up.

This produces a linker error if you try to use this function:
```
navfn_test.cpp:(.text+0x44): undefined reference to `navfn::create_nav_plan_astar(unsigned char const*, int, int, int*, int*, float*, int)'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

The const from the header is probably wrong, since the array will get modified.

DLu, 2018-09-05:
For reference: 
https://github.com/ros-planning/navigation/blob/c7bcd005fde62d3eaa2069b96f33ba4c472eadb8/navfn/src/navfn.cpp#L58

I'm not sure if the A* code actually works, so I would generally be wary of running it, but if you had a PR to fix the const issue I'd merge it.

ipa-fez, 2018-09-06:
https://github.com/ros-planning/navigation/pull/788

I ended up not using this function and am now using the NavFn class directly so I have no idea if the code works, but this should fix the signature mismatch.

mikeferguson, 2018-09-07:
Fixed in at least Melodic

closed 2018-09-07 by mikeferguson