ros-planning/navigation #793: runBehaviour() in recovery and costmap update

GMahmoud, 2018-09-13:
Hi guys,

I have worked recently on a rotate recovery behavior. So I tested your implemented plugin and I noticed that the the local costmap (obstacle layer only) is not updated when the behavior is running. This issue is caused by the ```while(n.ok()){ ... }``` loop running in the main thread. 
In my opinion, it is better to create a new std::thread to start the recovery behaivor and to avoid blocking the costmap updates for the forward simulation.

What do you think ?

Cheers
MG

mikeferguson, 2018-09-15:
That does seem reasonable -- my only concern would be if it introduces any race conditions (but from a quick look it appears that the recovery behavior does not alter the cost map, so I think we're OK).

casiarobot, 2020-06-22:
A case is that nobody can cancel the navigation task when in recovery behavior.  Especially, a situation u have to stop the robot. U can publish /cmd_vel to zero, but when stop it the robot will continue running rotate recovery behavior.

closed 2025-06-04 by mikeferguson