ros-planning/navigation #794: move_base Warning: Invalid argument "/map" passed to canTransform argument target_frame

oeyyey, 2018-09-17:
I do roslaunch move_base.launch then it does't work with repeat these warning
```
Warning: Invalid argument "/map" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /tmp/binarydeb/ros-melodic-tf2-0.6.3/src/buffer_core.cpp
Warning: Invalid argument "/map" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /tmp/binarydeb/ros-melodic-tf2-0.6.3/src/buffer_core.cpp
Warning: Invalid argument "/map" passed to canTransform argument target_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 134 in /tmp/binarydeb/ros-melodic-tf2-0.6.3/src/buffer_core.cpp
```
I have no idea what is cause this issue because I can run it on another PC.
Please help.

so this is my .launch file
```
<?xml version="1.0" ?>
<launch>
  <!-- Arguments -->
  <arg name="model" default="$(env MECANION_MODEL)" doc="model type [mecanum]"/>
  <arg name="cmd_vel_topic" default="/cmd_vel" />
  <arg name="odom_topic" default="odom" />

  <!-- move_base -->
  <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <param name="base_local_planner" value="dwa_local_planner/DWAPlannerROS" />
    <rosparam file="$(find mecanion_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find mecanion_navigation)/param/costmap_common_params_$(arg model).yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find mecanion_navigation)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find mecanion_navigation)/param/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find mecanion_navigation)/param/move_base_params.yaml" command="load" />
    <rosparam file="$(find mecanion_navigation)/param/dwa_local_planner_params_$(arg model).yaml" command="load" />
    <remap from="cmd_vel" to="$(arg cmd_vel_topic)"/>
    <remap from="odom" to="$(arg odom_topic)"/>
  </node>
</launch>
```

geerniya, 2018-09-21:
@oeyyey I also meet this issue in rviz. Have you solve it?
ROS kinetic, Ubuntu 16.04
`roslaunch turtlebot3_gazebo turtlebot3_world.launch`
`roslaunch turtlebot3_navigation turtlebot3_navigation.launch map_file:=$HOME/map.yaml`

[image]

And when I click the `2D Nav Goal` , the turtlebot do not move.

oeyyey, 2018-09-21:
@geerniya I solved by clone working version of Navigation package from my another computer.
So I upload it on my github (https://github.com/oeyyey/navigation-backup) 

First of all I solved warning by remove "/" in global_costmap_params.yaml and local_costmap_params.yaml but there is a problem in local costmap doesn't update itself. So I design to clone my Navigation package from my old computer then it work fine. 

PS. catkin_make the working version is required pcl_conversation but issue version is not.

ivan-r2d2, 2018-10-26:
I got the same error..
I solved it by removing the "/" in global_costmap_params.yaml and local_costmap_params.yaml as @oeyyey did

ivan-r2d2, 2018-10-31:
I also noticed that in ROS Melodic the 'move_base'  does not work well (the robot does a random motion and keeps rotating),,, but using the same code/parameters in ROS Kinetic works very well....
I think that this is due to the following issue
 
http://answers.gazebosim.org/question/20856/gazebo-9-ros-melodic-diff-robot-pluggin-rotates-robot-in-the-incorrect-direction/

jarvisschultz, 2018-12-07:
There are so many configs floating around that still have leading slashes that it makes me think it would be a good idea to patch `move_base` to automatically strip the leading slashes if they are there, and print a warning for users to remove them. Anyone have any insight as to whether a PR of that nature would be likely to be accepted?

SteveMacenski, 2018-12-07:
@jarvisschultz during the TF -> TF2 transition this was a decision that was made and I believe in the Melodic transition [TF1 was deprecated](http://wiki.ros.org/melodic/Migration) the code to handle leading slashes was _removed_ because it's no longer supported intentionally. The TF2 migration guide can be found [here](http://wiki.ros.org/tf2/Migration).

jarvisschultz, 2018-12-07:
@SteveMacenski thanks for answering quickly. I agree that all users should migrate their configurations to have no leading slashes, but even given the time to make the transition I've run into a few packages that have not done so. I guess if they want to release on melodic they'll be forced to update the configs.

DLu, 2019-01-14:
I'm marking this fixed as of #831 with a note for future visitors to this ticket to change your local config to not include leading slashes.

closed 2019-01-14 by DLu

CxyCH, 2019-07-04:
I have a question. When I use multi-robots and the frame name in the global_costmap_param.yaml is /robot1/base_footprint. I still have the problem even I remove the "/" and the name is "robot1/base_footprint". So I also need to remove the second "/"?
ubuntu 16.04  ros-kinetic

ho0-kim, 2019-09-14:
> I have a question. When I use multi-robots and the frame name in the global_costmap_param.yaml is /robot1/base_footprint. I still have the problem even I remove the "/" and the name is "robot1/base_footprint". So I also need to remove the second "/"?
> ubuntu 16.04 ros-kinetic

I'm also trying multi-robots, but have the same problem.