ros-planning/navigation #796: Error in fake_localization package: lookupTransform. Not enough time?

JavierLRH, 2018-09-29:
I have a problem with the package fake_localization.

When I set a new initial_pose in rviz, I get the error ""Failed to lookup transform!"":
[image]

This is my tf tree, I think that it's ok.
[image]

- I looking for a solution for the problem and I find this question with the same problem that me.
https://answers.ros.org/question/287587/lookuptransform-is-throwing-error/
The problem is that we need to wait for the transforms to become available.

- I need to wait for the tf at least 0.052 s

RESULTS: for map to base_link
Chain is: map -> odom -> base_link
Net delay     avg = 0.0262871: max = 0.0519673

- tf2 provides a tool that will wait until a transform becomes available.
[image]

Sorry for my English.
Regards and thank you.

guohefu, 2018-10-30:
> I have a problem with the package fake_localization.
>
> When I set a new initial_pose in rviz, I get the error ""Failed to lookup transform!"":
> [image]
>
> This is my tf tree, I think that it's ok. [...]

I have the same problem, has it been resolved?

JavierLRH, 2018-10-30:
Not yet.

guohefu, 2018-10-30:
I change the code to the following to solve:
baseInMap = m_tfBuffer->lookupTransform(base_frame_id_, global_frame_id_, ros::Time(0));

JavierLRH, 2018-10-30:
How do you compile the navigation package.
When I compile it with catkin the performance is much worse.

guohefu, 2018-10-30:
What is your platform (architecture, OS & version/distro)?

JavierLRH, 2018-10-30:
A Laptop with a Intel I5. The SO is Ubuntu 16.04 and the ROS version is Kinetic.
When I use the ROS navigation from source (apt-get install) the performance is good, but when I compile the Github version, the percentage of CPU usage rise to the 100% and I get warning messages from the package. 

Something similar to this: https://answers.ros.org/question/258139/navigation-stack-runs-slow-when-installed-from-source.
I try to change the flags but the issue is the same.

mikeferguson, 2018-11-15:
So, it looks like the code used to use ros::Time(0) for "latest transform", rather than the actual timestamp (before the port to TF2). I'd suggest trying ros::Time(0) -- if that works, please open a PR.

JavierLRH, 2018-11-15:
I just try the package with ros::Time(0) and works well.
I want to make the PR but I don't know how to do it. I must create a branch first?
Regards.

mikeferguson, 2018-11-15:
If you can confirm that ros::Time(0) works, I can handle the PR.

JavierLRH, 2018-11-15:
Hi, I made a video showing how the change works.
https://www.youtube.com/watch?v=itXlTo5CyPQ
Thanks

DLu, 2020-07-24:
Closed in #1017

commit 4dca4370 (2020-07-24): Fix #796 (#1017)

Use ros::Time(0) instead of timestamp in message so as not to fail to lookupTransform.
commit 3c4d04ed (2020-07-24): Fix #796 (#1017)

Use ros::Time(0) instead of timestamp in message so as not to fail to lookupTransform.
commit 34980f15 (2020-07-24): Fix #796 (#1017)

Use ros::Time(0) instead of timestamp in message so as not to fail to lookupTransform.
closed 2020-07-24 by DLu