ros-planning/navigation #805: fake_localization: Param delta_yaw, change in the behaviou

JavierLRH, 2018-10-22:
Hi everyone.

There is a change in the behaviour of the parameter delta_yaw in the last update.
[image]

In the old version the tf between map and odom looks like this:
This version works nice
[image]

In the new version the tf between map and odom looks like this:
[image]

Thanks.

mikeferguson, 2018-10-26:
Almost certainly, this is the same bug as fixed in other places here: https://github.com/ros-planning/navigation/pull/762 -- the setEuler function is the wrong one, needs to use setRPY.

corot, 2019-05-10:
This seems to be the underlying problem: https://github.com/ros/geometry2/issues/190

Replacing with setRPY fixes this package, but the library still wrong, ready to plague other progects