ros-planning/navigation #808: Obstacles can not be cleared in time.

sweetquiet, 2018-11-02:
Hello, I often encounter obstacles can not be cleared, especially when there are many people, but these obstacles will not be cleared after people leave, so it will not plan the path in time. I looked at the code, as if there is a maximum distance laser problem, how should this problem be modified to completely solve? Thanks very much

alfred31, 2018-11-02:
Try setting clearing to true in the obstacle layer

obstacles_layer:
 observation_sources: scan
 scan: {sensor_frame: ...,..., clearing: true, .....,...}

sweetquiet, 2018-11-02:
This my parameter is always true. In fact, the reason is that there is a problem with the maximum distance of the laser in the code. Many people have encountered it, but there is no perfect solution, so please ask the developer or everyone to fix it.

reinzor, 2018-11-02:
Try setting `inf_is_valid`, this will also shoot a ray if there is no reflection (e.g. large open spaces). See http://wiki.ros.org/costmap_2d/hydro/obstacles for more details.

closed 2018-11-05 by sweetquiet