ros-planning/navigation #812: fake localization initalpose was not working

salo120day, 2018-11-13:
I use navigation stack to my robot simulation,bur when i want use the "2D pose Estimate" button to set my robot ,terminal warn me :
[ WARN] [1542017987.602044741]: Frame ID of "initialpose" (map) is different from the global frame /map
[ WARN] [1542017987.602823922]: Failed to lookup transform!
and i add these in launch file:

but it dose't work ,some master tell me how can fix this problem

mikeferguson, 2018-11-15:
Let's try to resolve this in #796

closed 2018-11-15 by mikeferguson

salo120day, 2018-11-15:
> Hi, i have the same problem. Look at this post #796
Thanks, i have seen the Post#796 ,you mean the problem caused by the TF transform time delay？

salo120day, 2018-11-15:
> Let's try to resolve this in #796

Thanks,But The resolve way : change the code like 
baseInMap = m_tfBuffer->lookupTransform(base_frame_id_, global_frame_id_, ros::Time(0)),is it?