ros-planning/navigation #818: Compilation error solved by add_compile_options(-std=c++11)

AntaresZhang, 2018-12-05:
Seems like the tf2 dependency needs c++ 11 to compile correctly, solved this issue by adding 
----------------------------------------------
add_compile_options(-std=c++11)
----------------------------------------------
to  all  the CMakelists.txt of the packages with tf2 dependency
I got instruction  from this link:
https://github.com/ros/geometry2/issues/334

DLu, 2018-12-06:
Please post 
 * The error you were getting
 * Your operating system
 * Your version of ROS
 * Which branch of the navigation repo you are on
 * The command used for compiling

mikeferguson, 2018-12-07:
Seeing as our CI is passing on dev/PR builds, I imagine this actually some local config error (improper branch for your platform and/or rosdistro)

DLu, 2019-01-11:
Could not reproduce. I'm guessing there's something else in your pipeline that requires c++11

closed 2019-01-11 by DLu

synthaseatp, 2019-11-14:
@AntaresZhang Hey! I am a complete noob to this. Where do I have to add the line again, please? There are a lot of `CMakeLists.txt` files in `geometry2` package, and I ended up adding `add_compile_options(-std=c++11)` to every folder of the package of `geometry2` and the compilation still fails.

Thanks for your help.

synthaseatp, 2019-11-14:
@DLu It doesn't. This is a fresh install of ROS and so far I have two packages other than `geometry2`, and both of them compiled alright. 
I don't know how the issue flagging works, is it the same as tickets? Can this one be reopened, please?

synthaseatp, 2019-11-14:
@AntaresZhang @DLu 
Okay, it worked now.
The line had to be added to main `CMakeLists.txt` file in the `src` folder. I didn't try to write there fearing it is a "protected" file. A `sudo nano` took care of it.

Many thanks!