ros-planning/navigation #824: Assertion failure

randaz81, 2018-12-11:
I often get the following assertion failure when I start the module:

amcl: /usr/local/src/robot/ros-source/src/navigation/amcl/src/amcl/sensors/amcl_laser.cpp:315: static double amcl::AMCLLaser::LikelihoodFieldModel(amcl::AMCLLaserData*, pf_sample_set_t*): Assertion `pz <= 1.0' failed.

Are you familiar with this error?
I'm using melodic-devel branch.
I can provide you more details if requested.
Thank you

patrickcjh, 2019-12-29:
From the [wiki](http://wiki.ros.org/amcl):
> Note that whichever mixture weights are in use should sum to 1. The beam model uses all 4: z_hit, z_short, z_max, and z_rand. The likelihood_field model uses only 2: z_hit and z_rand. 

Ensure that your weights parameters does not add up to more than 1.

mikeferguson, 2020-03-29:
Appears this is a config error, not an actual bug. Closing. Please reopen if you find that there is a bug.

closed 2020-03-29 by mikeferguson