ros-planning/navigation #830: Should map_saver have raw mode ?

jingedawang, 2019-01-10:
There are many circumstances that raw map is needed, such as localization and path planning. But current version of map_saver can only save trinary map.

I am wondering if ROS organization have a plan to add this feature ? Or if I add this feature, are you willing to merge it ? I can guarantee backward compatibility.

jingedawang, 2019-01-11:
@mikeferguson

DLu, 2019-01-11:
I don't believe anyone is currently working on this. We are willing to merge pull requests.

jingedawang, 2019-01-14:
I have implemented it. Please have a look. #832

skpro19, 2021-06-16:
@jingedawang what do you mean by `trinary` map?

jingedawang, 2021-06-19:
> @jingedawang what do you mean by `trinary` map?

Free, occupied, unknown.

skpro19, 2021-06-20:
Makes sense.

closed 2025-06-04 by mikeferguson