ros-planning/navigation #834: move_base dies after setting a goal

abylikhsanov, 2019-01-15:
I am running ROS melodic on ASUS tinker board lubuntu. I have tested my navigation stack on regular Ubuntu but when I run the exact files on my embedded machine, as soon as I point a new goal, move_base dies and the debugger reports this:

```
    Thread 8 "move_base" received signal SIGSEGV, Segmentation fault.
    [Switching to Thread 0xb02fd410 (LWP 8259)]
    memset () at ../sysdeps/arm/memset.S:48
    48	../sysdeps/arm/memset.S: No such file or directory.
```

My kernel version is `4.4.71-ELAR-System`

closed 2019-01-17 by abylikhsanov

Kanishk598, 2021-09-01:
@abylikhsanov I'm having a very similar problem these days. As soon as I am setting a goal, I am getting this: `[move_base-3] process has died [pid 5414, exit code -11, cmd /opt/ros/melodic/lib/move_base/move_base __name:=move_base __log:=/home/kanishk/.ros/log/f0a6db88-0b56-11ec-8c3b-184f32c3a395/move_base-3.log]`
I am using ROS Melodic and a custom base_global_planner that I made (which was by the way working just fine before, no changes).
Any hints?

Kanishk598, 2021-09-03:
[This is the output of GDB debugger](https://github.com/Kanishk598/nav_stack/blob/Straight-line/gdb.txt), and [this is the output of Valgrind debugger](https://github.com/Kanishk598/nav_stack/blob/Straight-line/valgrind.txt). I am not able to check the mistake in the [planner that I've made](https://github.com/Kanishk598/nav_stack).