ros-planning/navigation #848: /media/youibot/U/new_corgi/src/robot_localizationunch

sweetquiet, 2019-01-28:
Hello, when I compile the navigation package, I always wait for a long time at move_base_node, about 10 minutes. What's the reason? Is there a good way? Thank you.

DLu, 2019-02-11:
This is a function of the speed of your computer and the amount of memory available. I investigated whether it was related to #800 but it did not seem to affect the runtime. Can you provide
 * What branch you are running
 * What operating system
 * Your compilation command
 * What step specifically it gets stuck on
 * What your memory usage looks like during compilation

sweetquiet, 2019-02-12:
First of all, thank you for your reply. I have encountered this situation on computers with different configurations. My current computer configuration is: 
1. Intel < Core < i7-7700HQ CPU @ 2.80GHz * 8.
2. ubuntu16.04 (64 bits).
3. Memory 16G.
4. Runs kinetic.
5. The compilation command is catkin_make.
6. When compilation is 100% progressing, it always stops at move_base_node.
7. Memory usage during compilation is 18% (16G).

sweetquiet, 2019-02-12:
Hello, David. Thank you very much. I added that link library in CmakeLists.txt of costmap_2d according to the [#800](https://github.com/ros-planning/navigation/pull/800) method you mentioned. It really works. I will close this problem. Thank you again.

closed 2019-02-12 by sweetquiet