ros-planning/navigation #852: dwa_local_planner: Proper scoring of turn in place "Trajectories"

jimmyw404, 2019-02-01:
I had some difficulty with a specific path challenge of:

1. Navigating to a goal in unexplored space
2. Entering an unexplored room
3. Seeing that there was no path through that room to the goal
4. Performing an in-place 180ish degree turn to navigate back out the doorway.

I'm new to move_base/navigation/dwa_local_planner so I'm probably missing something, but looking at the six critics responsible for selecting a trajectory that will cause the robot to turn in place, it appears that the goal_front_costs_, alignment_costs_ and goal_costs_ are all instances of MapGridCostFunction which all create a cost map from the current global plan and then score each trajectory by accumulating the cells that trajectory traverses. However in a turn in place "trajectory" isn't a true trajectory since it has no path, and so all the possible have the same cells and have the same score. This issue is shown in the below image where the robot was asked to navigate directly behind it, and it kind of wiggles around slowly.

[image]

Again, I'm new to this library, but I fixed by problem by adding a new TrajectoryCostFunction that scored based on the angular difference between the yaw at the end of the trajectory and the yaw of the target pose that is closest to the end of the trajectory. The code of the HeadingCostFunction::scoreTrajectory() function I wrote is below:

```
double HeadingCostFunction::scoreTrajectory(base_local_planner::Trajectory &traj)
{
	if(traj.getPointsSize() == 0)
	{
		ROS_ERROR("W: HeadingCostFunction::scoreTrajectory No trajectory when scoring, returning error value");
		return -1;
	}

	if(m_target_poses.size() == 0)
	{
		ROS_ERROR("W: HeadingCostFunction::scoreTrajectory No target poses when scoring, returning error value");
		return -2;
	}

	double endX,endY,endTh;
	traj.getEndpoint(endX,endY,endTh);

	geometry_msgs::PoseStamped tgtPose;
	tgtPose = m_target_poses.front();

	bool getClosest = true;
	int bestIdx = 0;
	double shortestDistSq = DBL_MAX;
	if(getClosest)
	{
		//OK, we want the tgtPose that is linearly interpolated between the closest two path points to it.
		//For now just get the closest to the trajectory endpoint.
		for(int i = 0; i < m_target_poses.size(); ++i)
		{
			geometry_msgs::PoseStamped pathPose = m_target_poses[i];
			Eigen::Vector2d delta(endX-pathPose.pose.position.x, endY-pathPose.pose.position.y);
			double distSq = delta.squaredNorm();
			if(distSq < shortestDistSq)
			{
				shortestDistSq = distSq;
				tgtPose = pathPose;
				bestIdx = i;
			}
		}
	}

    double tgtTh = tf2::getYaw(tgtPose.pose.orientation);

    double normDeltaTh = fabs(angles::shortest_angular_distance(tgtTh, endTh));
	return normDeltaTh;
}
```

This causes any score to value bringing the nose of the robot toward the target orientation.

Is there something I am missing about this problem? Maybe using the DWA local planner isn't the right choice or there are some tuning / features I don't have configured properly? Or are there consequences to the above approach I don't understand yet?

Thanks.

DLu, 2019-02-11:
This is an unfortunately common behavior when using DWA. Oftentimes the problem is circumvented by tuning the oscillation parameters such that it will continue to turn one way for long enough before switching back to the other turning direction so that it makes sense to keep turning that direction. 

You can also get turn in place behavior by setting the min_trans_vel to be negative, although that may have other consequences. 

There's nothing wrong with tuning the cost functions by hand, technically. However, DWA was not designed to handle all the use cases. If you like writing your own critics, I recommend [DWB](https://github.com/locusrobotics/robot_navigation/tree/master/dwb_local_planner)

closed 2019-02-11 by DLu

HappySamuel, 2021-07-29:
Hi @jimmyw404 

For your issue (4), I did some modifications to DWA local planner and Base local planner. My modified DWA local planner currently support to navigate the robot to the a goal that is opposite to its orientation. I have tested it in Gazebo and also on my real Robot.

I totally made 4 changes to both DWA local planner and Base local planner. The files I modified are:
(1) dwa_local_planner/include/dwa_local_planner/dwa_planner_ros.h
(2) dwa_local_planner/src/dwa_planner_ros.cpp
(3) base_local_planner/include/base_local_planner/latched_stop_rotate_controller.h
(4) base_local_planner/src/latched_stop_rotate_controller.cpp

You can refer to the changes of my fork of navigation stack:
https://github.com/HappySamuel/navigation
I have marked the changes i added/modified with //SAMUEL in those files.

Hope this can help your usage.

Best,
Samuel

jimmyw404, 2021-07-29:
Thanks, @HappySamuel ! I'll check it out.