ros-planning/navigation #856: costmap_2d::Costmap2DROS segmentation fault

FaboNo, 2019-02-06:
Dear All

I have a program which is working well on my laptop but which is creating a segmentation fault on an Intel NUC

system setup (both laptop and Intel NUC):
UBUNTU 16.04
ROS Kinetic
navigation stack 1.14.4 (intel NUC)
navigation stack: 1.14.3 (laptop)

I wrote a simple program (which is generation a seg fault):
#include <ros/ros.h>
#include <tf/transform_listener.h>
#include <costmap_2d/costmap_2d_ros.h>

    int main(int argc, char **argv) {
  	  ros::init(argc, argv, "hector_exploration_node");
	  ros::NodeHandle nh;
	  ros::NodeHandle nh_private("~");
	  tf::TransformListener tf(ros::Duration(10));
	  costmap_2d::Costmap2DROS costmap("global_costmap", tf);
  	  ros::spin();
  	  return 0;
   }

the yaml file for the costmap is : 

     global_costmap:
       map_type: costmap
       track_unknown_space: true
       unknown_cost_value: 255
       obstacle_range: 2.5
       raytrace_range: 3.0
       inflation_radius: 0.4
       footprint: [[0.32, 0.32],
              [-0.32, 0.32],
              [-0.32, -0.32],
              [0.32, -0.32]]
  
       transform_tolerance: 0.7
       inscribed_radius: 0.33
       circumscribed_radius: 0.35
       global_frame: /map
       robot_base_frame: /base_link
       update_frequency: 0.5
       publish_frequency: 0.5
       static_map: true
       rolling_window: false
       plugins: 
         - {name: static_map,       type: "costmap_2d::StaticLayer"}
         - {name: inflated_map,       type: "costmap_2d::InflationLayer"}

Using gdb I got the following information:
     
```
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/emeric/catkin_planning_ws/devel/lib/hector_exploration_node/hector_exploration_node 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffee6d6700 (LWP 4813)]
[New Thread 0x7fffeded5700 (LWP 4814)]
[New Thread 0x7fffed6d4700 (LWP 4815)]
[New Thread 0x7fffeced3700 (LWP 4821)]
[New Thread 0x7fffe7fff700 (LWP 4834)]

Thread 1 "hector_explorat" received signal SIGSEGV, Segmentation fault.
0x00007ffff7132dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
(gdb) backtrace
#0  0x00007ffff7132dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
#1  0x00007ffff7b82511 in costmap_2d::Costmap2DROS::Costmap2DROS(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, tf::TransformListener&) () from /opt/ros/kinetic/lib/libcostmap_2d.so
#2  0x000000000041bcdc in main (argc=1, argv=0x7fffffffd958) at /home/emeric/catkin_planning_ws/src/hector_navigation/hector_exploration_node/src/testCostmap.cpp:14
(gdb) backtrace full
#0  0x00007ffff7132dda in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
No symbol table info available.
#1  0x00007ffff7b82511 in costmap_2d::Costmap2DROS::Costmap2DROS(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, tf::TransformListener&) () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#2  0x000000000041bcdc in main (argc=1, argv=0x7fffffffd958) at /home/emeric/catkin_planning_ws/src/hector_navigation/hector_exploration_node/src/testCostmap.cpp:14
        nh = {namespace_ = "/", unresolved_namespace_ = "", remappings_ = std::map with 0 elements, unresolved_remappings_ = std::map with 0 elements, 
          callback_queue_ = 0x0, collection_ = 0x64a310, ok_ = true}
        nh_private = {namespace_ = "/hector_exploration_node", unresolved_namespace_ = "/hector_exploration_node", remappings_ = std::map with 0 elements, 
          unresolved_remappings_ = std::map with 0 elements, callback_queue_ = 0x0, collection_ = 0x64b3a0, ok_ = true}
        tf = <incomplete type>
        costmap = {layered_costmap_ = 0x0, name_ = "global_costmap", tf_ = @0x7fffffffd380, global_frame_ = "map", robot_base_frame_ = "base_link", 
          transform_tolerance_ = 0.29999999999999999, map_update_thread_shutdown_ = false, stop_updates_ = false, initialized_ = true, stopped_ = false, 
          robot_stopped_ = false, map_update_thread_ = 0x0, timer_ = {impl_ = {px = 0x0, pn = {pi_ = 0x0}}}, 
          last_publish_ = {<ros::TimeBase<ros::Time, ros::Duration>> = {sec = 0, nsec = 0}, <No data fields>}, publish_cycle = {<ros::DurationBase<ros::Duration>> = {
              sec = 0, nsec = 0}, <No data fields>}, plugin_loader_ = <incomplete type>, old_pose_ = {<tf::Transform> = {m_basis = {m_el = {{m_floats = {1, 0, 0, 0}}, 
                  {m_floats = {0, 1, 0, 0}}, {m_floats = {0, 0, 1, 0}}}}, m_origin = {m_floats = {1e+30, 1e+30, 1e+30, 0}}}, 
            stamp_ = {<ros::TimeBase<ros::Time, ros::Duration>> = {sec = 0, nsec = 0}, <No data fields>}, frame_id_ = "NO_ID_STAMPED_DEFAULT_CONSTRUCTION"}, 
          publisher_ = 0x0, dsrv_ = 0x0, configuration_mutex_ = {m = {__data = {__lock = 0, __count = 0, __owner = 0, __nusers = 0, __kind = 1, __spins = 0, 
                __elision = 0, __list = {__prev = 0x0, __next = 0x0}}, __size = '\000' <repeats 16 times>, "\001", '\000' <repeats 22 times>, __align = 0}}, 
          footprint_sub_ = {impl_ = {px = 0x0, pn = {pi_ = 0x0}}}, footprint_pub_ = {impl_ = {px = 0x0, pn = {pi_ = 0x0}}}, 
          unpadded_footprint_ = std::vector of length 0, capacity 0, padded_footprint_ = std::vector of length 0, capacity 0, footprint_padding_ = 0, old_config_ = {
            groups = {transform_tolerance = 6.9533558072958799e-310, update_frequency = 6.953355807314852e-310, publish_frequency = 6.9533558072713742e-310, 
              width = -11600, height = 32767, resolution = 3.24956461330188e-317, origin_x = 1.0869444208507424e-322, origin_y = 6.9533491725705344e-310, 
              footprint = "", robot_radius = 3.2503037355080585e-317, footprint_padding = 3.2503037355080585e-317, state = true, name = "Default"}, 
            transform_tolerance = 3.250414406212727e-317, update_frequency = 2.5622244393327046e-320, publish_frequency = 6.9533491588572483e-310, width = -219960880, 
            height = 32767, resolution = 6.9533449472484047e-310, origin_x = 6.9533490667912279e-310, origin_y = 2.5622244393327046e-320, footprint = "", 
            robot_radius = 6.9533490667101023e-310, footprint_padding = 3.24956461330188e-317}}
```
Can you tell me please why I got segmentation fault here because I am clueless.

Thank you very much for your help

FaboNo, 2019-02-06:
After more investigation, I  modified the yaml file:

         global_costmap:
          footprint: [[0.32, 0.32],
              [-0.32, 0.32],
              [-0.32, -0.32],
              [0.32, -0.32]]
  
         transform_tolerance: 0.7
         inscribed_radius: 0.33
         circumscribed_radius: 0.35
         global_frame: map
         robot_base_frame: base_link
         update_frequency: 1
         publish_frequency: 1
         static_map: true
         rolling_window: false
         plugins: 
             - {name: static_map,       type: "costmap_2d::StaticLayer"}
             - {name: inflated_map,       type: "costmap_2d::InflationLayer"}

        static_map:
             map_topic:/map
             unknown_cost_value: 255
             subscribe_to_updates:true
             track_unknown_space: true

        inflated_map:
            inflation_radius: 0.4

And at least in the debug, I can progress:
```
Thread 1 "hector_explorat" hit Breakpoint 1, main (argc=1, argv=0x7fffffffd958)
    at /home/emeric/catkin_planning_ws/src/hector_navigation/hector_exploration_node/src/testCostmap.cpp:14
14		costmap_2d::Costmap2DROS costmap("global_costmap", tf);
(gdb) s
[ERROR] [1549469721.506457032]: poll failed with error Interrupted system call
[ INFO] [1549469721.698410311]: Using plugin "static_map"
[ INFO] [1549469721.728158733]: Requesting the map...
[ INFO] [1549469721.934175004]: Resizing costmap to 2048 X 2048 at 0.050000 m/pix
[ INFO] [1549469722.031948733]: Received a 2048 X 2048 map at 0.050000 m/pix
[ INFO] [1549469722.042226929]: Using plugin "inflated_map"
[New Thread 0x7fffe5eb9700 (LWP 11606)]
[New Thread 0x7fffe56b8700 (LWP 11632)]
[ERROR] [1549469722.081695058]: No Transform available Error looking up robot pose: "[\?z?k" passed to lookupTransform argument source_frame does not exist. 

16	  	ros::spin();
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) n
Program not restarted.
(gdb) n
[ERROR] [1549469737.289766118]: poll failed with error Interrupted system call

Thread 8 "hector_explorat" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe56b8700 (LWP 11632)]
0x00007ffff712e4a1 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
(gdb) backtrace full
#0  0x00007ffff712e4a1 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
No symbol table info available.
#1  0x00007ffff7b7cc4e in costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#2  0x00007ffff7b7debf in costmap_2d::Costmap2DROS::updateMap() () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#3  0x00007ffff7b7f056 in costmap_2d::Costmap2DROS::mapUpdateLoop(double) () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#4  0x00007ffff639c5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
No symbol table info available.
#5  0x00007ffff61756ba in start_thread (arg=0x7fffe56b8700) at pthread_create.c:333
        __res = <optimized out>
        pd = 0x7fffe56b8700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737042417408, -5771018804834229841, 0, 140737488338367, 140737042418112, 13456984, 5771066259907125679, 
                5771033994774421935}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"

```
It looks like there a problem with the transform but the transformation between map and base_link is working:
```
emeric@emeric-desktop:~$ rosrun tf tf_echo map base_link
At time 1549468483.364
- Translation: [-0.008, -0.030, 0.140]
- Rotation: in Quaternion [-0.002, -0.022, -0.016, 1.000]
            in RPY (radian) [-0.003, -0.044, -0.032]
            in RPY (degree) [-0.158, -2.500, -1.848]
At time 1549468484.017
- Translation: [-0.007, -0.035, 0.160]
- Rotation: in Quaternion [-0.002, -0.022, -0.016, 1.000]
            in RPY (radian) [-0.003, -0.044, -0.031]
            in RPY (degree) [-0.163, -2.503, -1.799]
At time 1549468485.041

```
The error seems to be triggered by this piece of code in costmap_2d_ros.cpp:
```
try {
    tf_.transform(robot_pose, global_pose, global_frame_);
  } catch (tf2::LookupException & ex) {
    RCLCPP_ERROR(get_logger(),
      "No Transform available Error looking up robot pose: %s\n", ex.what());
return false;
```
does somebody see the issue?
Thank you

DLu, 2019-02-11:
Typically this sort of thing is handled on answers.ros.org. Can you please post the details over there, including the CMakeLists you are using to compile your node?

DLu, 2019-02-11:
(and post a link to the question here to complete the loop)

closed 2019-02-11 by DLu