ros-planning/navigation #876: does move_base still support actions?

nathangeorge1, 2019-03-11:
In the [move_base tutorial](https://wiki.ros.org/move_base), the move_base/goal topic redirects to a [404 page](https://docs.ros.org/api/move_base_msgs/html/msg/MoveBaseActionGoal.html).

Also, it seems that `rostopic info /move_base/goal` on Kinetic gives `Type: geometry_msgs/PoseStamped`.

Does move_base support actions?

mikeferguson, 2019-03-11:
Yes, it does still support actions -- I'm not sure why the documentation is broke (that seems like a ROS.org issue, since that linking occurs automatically). You can see the definitions here: https://github.com/ros-planning/navigation_msgs

The move_base/goal topic has a PoseStamped within it, so maybe that is what you're seeing with rostopic info.

nathangeorge1, 2019-03-11:
O ok; so even though https://github.com/ros-planning/navigation_msgs/blob/jade-devel/move_base_msgs/action/MoveBase.action says Jade and uses PoseStamped, it will still work for ROS action. Ok my mistake.

closed 2019-03-11 by nathangeorge1