ros-planning/navigation #880: obstacle_layer has high delay for marking moving object using 2d lidar

SeanGuoXuan, 2019-03-23:
I am using static layer and obstacle layer for local costmap. I observe there are some delays(about 2s+) on local costmap reflecting moving object. Those green dots represent 2d lidar detected obstacles. However, costmap takes a few seconds to catch up with the real obstacles. The picture below shows this issue:
[image]

Here are config files:

1. costmap common 

`footprint: [[-0.3, -0.305], [-0.3, 0.305], [0.3, 0.305], [0.3, -0.305]]
footprint_padding: 0.05
robot_base_frame: base_link
update_frequency: 10.0
publish_frequency: 2.0
transform_tolerance: 0.2
resolution: 0.05
map_type: costmap
#layer definitions
static:
    enabled: true
    map_topic: /map
    lethal_cost_threshold: 55
    subscribe_to_updates: true
obstacles_laser:
    enabled: true
    obstacle_range: 10.0
    raytrace_range: 3.0
    inflation_radius: 0.5
    track_unknown_space: true
    observation_sources: scan
    scan: {sensor_frame: rplidar_link, data_type: LaserScan, topic: scan, marking: true, clearing: true}`

2. costmap local:

`global_frame: map
robot_base_frame: base_link
rolling_window: true
static_map: false
plugins:
    - {name: static,              type: "costmap_2d::StaticLayer"}
    - {name: obstacles,           type: "costmap_2d::ObstacleLayer"}`

What is causing this delay? Any idea?

SeanGuoXuan, 2019-03-26:
sorry, configuration mistake. Close it.

closed 2019-03-26 by SeanGuoXuan

artemiialessandrini, 2020-04-15:
If figured out, please, answer your question yourself before closing.

rvbabera, 2024-08-28:
Hi,
I am also having same issue with ros noetic please let me know how you fixed it ?