ros-planning/navigation #883: Costmap_2d segmentation fault

eneiva, 2019-04-08:
I had a strange behaviour running the **`mbf_costmap_nav and move_base`** which strongly indicates that there is something weird with **`costmap_2d`**. Whenever costmap_2d is called on a virtual machine I have no issues and it is working as intended, however when I run it from my embedded system, **`costmap_2d`** and nodes that uses it returns instantly a segmentation fault. All the other nodes/packages run seamlessly on both systems(embedded system and VM), there aren't presumably networking errors as the other packages are working. `roswtf` gives me nothing.

Both systems are running similar version of ubuntu 16.04, the virtual machine has a x86_64 distro meanwhile the embedded system runs on AARCH64.

I have thought about what could be so different from both systems but unfortunately I haven't reached any conclusion.

**Output from the terminal of the embedded system:**

The code ran as an example is: **`rosrun mbf_costmap_nav mbf_costmap_nav`** (self compiled)

```
nvidia@nvidia:~/catkin_ws$ rosrun --prefix 'gdb -ex run --args' mbf_costmap_nav mbf_costmap_nav 
GNU gdb (Ubuntu 7.11-0ubuntu1) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/nvidia/catkin_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav...done.
Starting program: /home/nvidia/catkin_ws/devel/lib/mbf_costmap_nav/mbf_costmap_nav 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb44a1120 (LWP 5690)]
[New Thread 0x7fb3ca1120 (LWP 5691)]
[New Thread 0x7fb34a1120 (LWP 5692)]
[New Thread 0x7fb2ca1120 (LWP 5693)]
[New Thread 0x7fb24a1120 (LWP 5694)]
[New Thread 0x7fb1ca1120 (LWP 5695)]
[New Thread 0x7fb14a1120 (LWP 5696)]
[New Thread 0x7fb0ca1120 (LWP 5697)]
[ INFO] [1554713705.534366709]: Using plugin "obstacle_layer"
[ INFO] [1554713705.602884899]:     Subscribed to Topics: 
[ INFO] [1554713705.662088021]: Using plugin "inflation_layer"
[New Thread 0x7f93fff120 (LWP 5698)]
[New Thread 0x7f937ff120 (LWP 5699)]

Thread 11 "mbf_costmap_nav" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f937ff120 (LWP 5699)]
0x0000007fb7b99674 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
(gdb) bt full
#0  0x0000007fb7b99674 in ros::Time::now() () from /opt/ros/kinetic/lib/librostime.so
No symbol table info available.
#1  0x0000007fb718fc98 in costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&) const () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#2  0x0000007fb7190db8 in costmap_2d::Costmap2DROS::updateMap() () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#3  0x0000007fb7191a34 in costmap_2d::Costmap2DROS::mapUpdateLoop(double) () from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#4  0x0000007fb6e0ecf8 in ?? () from /usr/lib/aarch64-linux-gnu/libboost_thread.so.1.58.0
No symbol table info available.
#5  0x0000007fb7039fb4 in start_thread (arg=0x7fb6e0ec30) at pthread_create.c:335
        pd = 0x7fb6e0ec30
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {547935482144, 549755803736, 0, 549755803727, 548529040432, 548531462144, 548485736176, 7754544, 1, 547935482144, 547935480080, 13568342539068514507, 
                0, 13568342538606892639, 4617315517961601024, 4591870180066957722, 4576918229304087675, 0, 0, 0, 0, 0}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, 
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#6  0x0000007fb78f9390 in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:89
No locals.
(gdb) s
Single stepping until exit from function _ZN3ros4Time3nowEv,
which has no line number information.
[ERROR] [1554713715.733687113]: poll failed with error Interrupted system call
[Thread 0x7fb34a1120 (LWP 5692) exited]
[Thread 0x7fb3ca1120 (LWP 5691) exited]
[Thread 0x7f937ff120 (LWP 5699) exited]
[Thread 0x7f93fff120 (LWP 5698) exited]
[Thread 0x7fb0ca1120 (LWP 5697) exited]
[Thread 0x7fb14a1120 (LWP 5696) exited]
[Thread 0x7fb1ca1120 (LWP 5695) exited]
[Thread 0x7fb24a1120 (LWP 5694) exited]
[Thread 0x7fb2ca1120 (LWP 5693) exited]
[Thread 0x7fb44a1120 (LWP 5690) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
 
```

The code ran as an example is: **`rosrun move_base move_base`** (std package)

```
nvidia@nvidia:~/catkin_ws$ rosrun --prefix 'gdb -ex run --args' move_base move_base
GNU gdb (Ubuntu 7.11-0ubuntu1) 7.11
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/kinetic/lib/move_base/move_base...(no debugging symbols found)...done.
Starting program: /opt/ros/kinetic/lib/move_base/move_base 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fb4c1d120 (LWP 4881)]
[New Thread 0x7faffff120 (LWP 4882)]
[New Thread 0x7faf7ff120 (LWP 4883)]
[New Thread 0x7faefff120 (LWP 4884)]
[New Thread 0x7fae7ff120 (LWP 4885)]
[New Thread 0x7fadfff120 (LWP 4886)]
[New Thread 0x7fad7ff120 (LWP 4887)]

Thread 1 "move_base" received signal SIGSEGV, Segmentation fault.
0x0000007fb7cd9b6c in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const
    () from /opt/ros/kinetic/lib/librostime.so
(gdb) bt full
#0  0x0000007fb7cd9b6c in ros::TimeBase<ros::Time, ros::Duration>::operator+(ros::Duration const&) const () from /opt/ros/kinetic/lib/librostime.so
No symbol table info available.
#1  0x0000007fb79110c8 in costmap_2d::Costmap2DROS::Costmap2DROS(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, tf::TransformListener&) ()
   from /opt/ros/kinetic/lib/libcostmap_2d.so
No symbol table info available.
#2  0x0000007fb7f455d4 in move_base::MoveBase::MoveBase(tf::TransformListener&) ()
   from /opt/ros/kinetic/lib/libmove_base.so
No symbol table info available.
#3  0x00000000004056b0 in main ()
No symbol table info available.
(gdb) s
Single stepping until exit from function _ZNK3ros8TimeBaseINS_4TimeENS_8DurationEEplERKS2_,
which has no line number information.
Unable to fetch general registers.: No such process.
Unable to fetch general registers.: No such process.
Unable to fetch general registers.: No such process.
(gdb) [Thread 0x7fad7ff120 (LWP 4887) exited]
[Thread 0x7fadfff120 (LWP 4886) exited]
[Thread 0x7fae7ff120 (LWP 4885) exited]
[Thread 0x7faefff120 (LWP 4884) exited]
[Thread 0x7faf7ff120 (LWP 4883) exited]
[Thread 0x7faffff120 (LWP 4882) exited]
[Thread 0x7fb4c1d120 (LWP 4881) exited]

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
```

What could I be missing? Are there any library that could be causing this seg fault?

This issue can be traced back from here: https://github.com/magazino/move_base_flex/issues/122.

https://github.com/ros-planning/navigation/issues/856 , I couldn't find any follow up to the later.

mikeferguson, 2019-04-09:
Yes, ABI compatability is unfortunately not well managed in ROS.

artemiialessandrini, 2020-07-08:
If I'm not mistaken, something is up with inflater_layer plugin, running [this snippet](http://wiki.ros.org/move_base_flex/Tutorials/SimpleSmachForMoveBaseFlex),  
and getting till `RECOVERY` state, when `clear_costmap` recovery looks like this:  
```
clear_costmap:  
  reset_distance: 0.0  
  force_updating: true  
  layer_names: [obstacles, inflater_layer]  
```  
Recieving this output:  
```[ INFO][7ffff7f94780][/move_base_flex/AbstractNavigationServer::callActionRecovery:273]: Start action "recovery" using recovery "clear_costmap" of type "clear_costmap_recovery/ClearCostmapRecovery"
[New Thread 0x7fffb37fe700 (LWP 23057)]
[New Thread 0x7fffb2ffd700 (LWP 23058)]
[ WARN][7fffb2ffd700][/move_base_flex/ClearCostmapRecovery::runBehavior:99]: Clearing both costmaps to unstuck robot (0.00m).

Thread 18 "mbf_costmap_nav" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffb2ffd700 (LWP 23058)]
__GI___pthread_mutex_lock (mutex=0x3fa99999a0000000) at ../nptl/pthread_mutex_lock.c:65
65	../nptl/pthread_mutex_lock.c: No such file or directory.
```  
Hopefully, It's going to be helpful.  
Also, when `layer_names: [obstacles]`, things are working with no issue and clear_costmap is doing it's job.