ros-planning/navigation #885: footprint_pub and footprint_sub are different message types on same topic

mkhansen-intel, 2019-04-17:
@DLu or @mikeferguson - While running in ROS2 we've noticed that the "footprint" topic has a different publisher message type (PolygonStamped) than subscriber type (Polygon). This doesn't work at all with the Opensplice RMW layer implementation (it causes the costmap nodes to crash). Looking at the code, I don't understand why the two are different. I tried changing them to the same (PolygonStamped) but that caused the costmap inflations to be wrong. 

Could you explain the intent of the publisher and subscriber being different message types on the same topic? We may need a different way to do this in ROS2. 
cc: @crdelsey

mikeferguson, 2019-04-17:
Can you give a pointer to where the subscriber is created? I'm not sure what that is all about.

The publisher I believe is mainly for visualization purposes -- so I don't think they are intended to be connected together (which sounds like what you are trying to do?)

mkhansen-intel, 2019-04-17:
The subscriber is created here:
https://github.com/ros-planning/navigation/blob/b52e0ea9362131dc81326da471792ab8698c9564/costmap_2d/src/costmap_2d_ros.cpp#L138
```
 // subscribe to the footprint topic
  std::string topic_param, topic;
  if (!private_nh.searchParam("footprint_topic", topic_param))
  {
    topic_param = "footprint_topic";
  }

  private_nh.param(topic_param, topic, std::string("footprint"));
  footprint_sub_ = private_nh.subscribe(topic, 1, &Costmap2DROS::setUnpaddedRobotFootprintPolygon, this);

  if (!private_nh.searchParam("published_footprint_topic", topic_param))
  {
    topic_param = "published_footprint";
  }

  private_nh.param(topic_param, topic, std::string("oriented_footprint"));
  footprint_pub_ = private_nh.advertise<geometry_msgs::PolygonStamped>("footprint", 1);
```
Then later the callback is declared, note that it is expecting a Polygon msg, not PolygonStamped:
```
void Costmap2DROS::setUnpaddedRobotFootprintPolygon(const geometry_msgs::Polygon& footprint)
{
  setUnpaddedRobotFootprint(toPointVector(footprint));
}
```
So I'm not sure what the intent is here, are they supposed to be on 2 separate topics?

mikeferguson, 2019-04-17:
> So I'm not sure what the intent is here, are they supposed to be on 2 separate topics?

I think so, @DLu might chime in differently (he's probably more familiar with this piece of code than I am). I think the PolygonStamped publisher is there mainly for visualization (I don't think you can visualize a pure Polygon message in RVIZ, since it has no timestamp or header to align the data).

Since they are different messages, the subscriber shouldn't get any messages from the publisher in ROS1.

mkhansen-intel, 2019-04-17:
So a couple questions, if the subscriber doesn't get messages from the publisher, is something else supposed to publish to that topic so that the subscriber does get a callback? Also, if they're not meant to communicate, shouldn't they be on separate topics?

Sorry for so many questions, I'm still struggling a bit with the intent here. Is this to sync the local and global costmaps?

mikeferguson, 2019-04-17:
So, these aren't features I use much/often, but here's my understanding of these two completely separate features:

1) The publisher is meant for visualization in RVIZ. This has been there since the beginning of time in the navigation stack, and is the outline you often see in RVIZ screen captures of robots navigating.

2) The subscriber exists for external programs to change the footprint. For instance, say your robot just attached itself to a cart. The cart presumably has a larger footprint the robot, so now you have to change your footprint for future plans (or you'll run the cart into things). Once you detach the cart, you need to reset the footprint back to the original size. The logic of what the proper footprint is (and all the cart attaching/detaching) exists externally to the navigation stack. I think this feature was added quite a bit later (if I'm not mistaken it was added when some folks had the PR2 moving carts around and doing dish collection at WG), and so people probably didn't notice the topic names were the same. This code has also been refactored multiple times (first for the split of costmaps into plugins, which @DLu did as an intern at WG, and then a second time when @DLu removed the "footprint" layer and moved the functionality into the base costmap class)

mkhansen-intel, 2019-04-17:
Thanks so much for the explanation. We've been viewing the published topic in RVIZ, so that makes sense, I just  didn't understand why the subscriber was there, and why they had different types. From your explanation above, it sounds to me like they should be separate topics. The first one "footprint" would be what is published to RVIZ (same as now), but the second one could be a different topic such as "dynamic_footprint" or something similar, for making changes to the footprint such as when a cart is attached / detached. Let me know if you know of any reason why that wouldn't work, meanwhile I'm going to make that change and test it.

mikeferguson, 2019-04-18:
Really, the "dynamic_footprint" subscriber should probably be a service or action interface. In theory, you really want to know that the footprint has been updated before you send it another goal. In your ROS2 setup, I imagine that said action interface might even be something that you want the behavior trees to be able to call (for instance, you might have a long running task that wants to go to location A, pick up a cart, go to B, drop off the cart. You'd have to change the footprint as part of the task).

mkhansen-intel, 2019-04-18:
Does this also get used by robots like the PR2 when they move their arms? For example if the PR2 sticks it's arm out sideways, it will be reflected as a wider footprint?

mikeferguson, 2020-03-03:
We've half-patched this in #947 - default topic name will get updated in noetic.

closed 2020-03-03 by mikeferguson