ros-planning/navigation #895: Local Costmap showing some objects but not all of them

joekeo, 2019-06-24:
Hello,

Sorry if this is a double post, i can seem to find anyone with the same issue.
I have been working with move_base for a while now using different platforms. But I have never experienced this issue, maybe some of you did.

The local costmap seems to only be showing some of the obstacles but not all of them, particularly dynamic ones.
The picture below shows an exmaple, where the local cost map picks up objects by the side of the robot, but not the table that is just infront of him.

[image]

Similar things happen with dynamic ones (people) the costmap shows them at random points around the robot, but not all the time.

The robot is setup with a Hokuyo lidar. here are the config files:
common_costmap.yaml
```
global_frame: map
robot_base_frame: base_link

footprint:  [[0.42, 0.26], [-0.42, 0.26], [-0.42, -0.26], [0.42, -0.26]]
footprint_padding: 0.03

inflater_layer:
  inflation_radius: 0.05
  cost_scaling_factor: 0.5

plugins:
- {name: obstacles_layer, type: "costmap_2d::ObstacleLayer"}
- {name: inflater_layer, type: "costmap_2d::InflationLayer"}
```
global_costmap.yaml:
```
global_costmap:
  update_frequency: 5.0
  static_map: true
  rolling_window: false
  resolution: 0.05
  static_layer:
    map_topic: /map
    subscribe_to_updates: true
    lethal_cost_threshold: 60
  inflater_layer:
    inflation_radius: 0.25
    cost_scaling_factor: 0.5

  plugins:
  - {name: static_layer, type: "costmap_2d::StaticLayer"}
  - {name: inflater_layer, type: "costmap_2d::InflationLayer"}
```
local_costmap.yaml:
```
local_costmap:
  update_frequency: 5.0
  publish_frequency: 5.0
  static_map: false
  rolling_window: true
  width: 8.0
  height: 8.0
  resolution: 0.05
  

  obstacles_layer:
    observation_sources: laser_scan_sensor
    laser_scan_sensor: {sensor_frame: horizontal_laser_link, data_type: LaserScan, topic: scan, clearing: true, marking: true}
    max_obstacle_height: 3
    obstacle_range: 10
    raytrace_range: 11
```
If anyone have any Idea why this is happening, you will help me sleep at night.

Best,
Pepe

DLu, 2019-06-25:
Do you have an answers.ros.org link?

joekeo, 2019-06-25:
Good Idea, just did that, lets see what happens. Thanks!

DLu, 2019-06-26:
Closing in favor of http://answers.ros.org/question/326867/local_costmap-not-showing-every-obstacle/

closed 2019-06-26 by DLu