ros-planning/navigation #898: Inflation layer of costmap is cleared by robot footprint

junmingtan, 2019-07-02:
The inflated regions of the costmap is being cleared upon traversal by the robot.

[image]

Is this an intended behaviour and if so, how can i disable this behaviour. I wish to retain all inflated regions to ensure that the global path planned is always consistent and safe.

The parameters used are as follows:

- costmap_common_params.yaml
`obstacle_range: 12.0
raytrace_range: 12.5

robot_radius: 0.17

obstacle_layer:
  enabled: true
  max_obstacle_height: 1.2
  min_obstacle_height: 0.0
  obstacle_range: 3.0
  raytrace_range: 12.0
  combination_method: 1
  observation_sources: laser_scan_sensor 
  track_unknown_space: true
  footprint_clearing_enabled: false

  laser_scan_sensor:
    data_type: LaserScan
    topic: /scan
    marking: true
    clearing: true 
    expected_update_rate: 0
    min_obstacle_height: 0.05
    max_obstacle_height: 0.30

inflation_layer:
  enabled:              true
  cost_scaling_factor:  2.0  # exponential rate at which the obstacle cost drops off (default: 10)
  inflation_radius:     0.5 # max. distance from an obstacle at which costs are incurred for planning paths.

map_layer:
  enabled:              true
  map_topic:            map
  trinary_costmap:      true
  use_maximum:          true
  first_map_only:       true

boundary_layer:
  enabled:                      true
  map_topic:                    user_map
  trinary_costmap:              true
  use_maximum:                  true
  subscribe_to_updates:         true
  first_map_only:               true`

- global_costmap.yaml
`global_costmap:
  global_frame: map
  robot_base_frame: base_footprint

  update_frequency: 10.0
  publish_frequency: 10.0
  transform_tolerance: 1.5

  static_map: true
  map_type: costmap
  plugins:
   - {name: boundary_layer,     type: "costmap_2d::StaticLayer"}
   - {name: map_layer,          type: "costmap_2d::StaticLayer"}
   - {name: inflation_layer,    type: "costmap_2d::InflationLayer"}
    `

EwingKang, 2019-07-09:
Not sure if you're talking about the parameter "footprint_clearing_enabled"(http://wiki.ros.org/costmap_2d/hydro/obstacles), but can you provide a higher resolution explanation of your problem? I couldn't see any issue with your gif

junmingtan, 2019-07-09:
The problem that I had is that the robot is erasing regions of the costmap that it has navigated across. This is found to be an intended behaviour and is solved by pull request #897

closed 2019-07-09 by junmingtan