ros-planning/navigation #907: Issue with outlineMap

rittersb, 2019-07-25:
I may be using the global_planner in a more unusual way, but it somewhat confused me when I switched from NavfnROS to global_planner and I saw a border around the costmap which is drawn in [outlineMap](https://github.com/ros-planning/navigation/blob/d99320a5ff8c05f3a6246810b044ccbd00524107/global_planner/src/planner_core.cpp#L287). To give a small bit of background, I plan in the local costmap, and do not use a global costmap because I don't know how far I will be traveling nor do I have prior knowledge of the area. This border messes up navigation when I command past the outline or outside the local costmap (because of the outline). In my use case I just commented out the code, but I thought it would be nice to make this optional through a parameter.

DLu, 2019-07-31:
I think having a parameter that optionally turned `outlineMap` off would be a fine idea for a PR.

sumejko92, 2019-12-03:
@DLu There is now a PR #926 regarding this.

closed 2020-02-13 by mikeferguson